paparazzi/paparazzi #2284: XBEE_API not working in Fixedwing - LPC21

dewagter, 2018-06-28:


dewagter, 2018-06-28:
- hardware-debugged the wait function: it works exactly as expected

dewagter, 2018-06-28:
- hardware debugged: when sending "+++" config string, the put_bytes is called 3 times but only the first "+" is actually sent over the serial. The XBee obviously never replies.

dewagter, 2018-06-28:
- while the XBee is being configured, there is also NO PWM so it looks like some master interrupt function is still off.

gautierhattenberger, 2018-06-28:
This should be checked, but maybe the `xbee_init` function is called (from the `modules_init`) before the `mcu_int_enable` function. It does nothing on stm32, but for lpc it means the uart interrupts won't be called, so nothing will happen after the first character.

dewagter, 2018-06-28:
See https://github.com/paparazzi/paparazzi/pull/2285

closed 2018-06-28 by dewagter