paparazzi/paparazzi #2292: [Install] Missing dependencies on fresh install

kirkscheper, 2018-07-10:
After a fresh (like spanking new) install of paparazzi, start.py fails with 'Import Error: no module named pygtk'. Seems like that 'python-gtk2-dev' is not properly set as a dependency for paparazzi-dev.

gautierhattenberger, 2018-12-18:
is it really `python-gtk2-dev` that is needed ? I don't have it and can still run `start.py`. I have `python-gtk2`, do you think it is enough to add this one ?

gautierhattenberger, 2018-12-18:
Apparently the `python-gtk2` is installed when recommended packages are installed as well (as it is a recommended package of python-matplotlib which is recommended by paparazzi-dev). I can add it to the dependency list anyway.

kirkscheper, 2018-12-18:
Yup, tested on a fresh install on a usb, working! Thanks

closed 2018-12-18 by kirkscheper