paparazzi/paparazzi #2294: bebop2_opticflow segfault at takeoff

tomvand, 2018-07-16:
Tested with commit da3e040

When taking off with the example bebop2_opticflow aircraft, the autopilot crashes with a segfault. The segfault appears to occur only after takeoff (during the initial climb). The segfault can also be triggered by manually lifting the drone from the ground (with motors killed. It may take a few attempts to trigger the crash). Just unkilling the motors on the ground does not seem to trigger the segfault.

The example bebop2 aircraft without optical flow does not run into a segfault. I suspect this is an optical flow problem.

tomvand, 2018-07-16:
I can replicate this crash with commit 30ab12 but not with a628e6 (the one before that), so that's when something went wrong it seems.

guidoAI, 2018-07-17:
I could not reproduce the problem, although I tried... The mentioned airframe was indeed not working well though. See the pull request: https://github.com/paparazzi/paparazzi/pull/2297

tomvand, 2018-07-18:
I tried your pull request #2297, but it does not fix the segfault. I'll try your derotation_threshold branch next.

tomvand, 2018-07-18:
Not a 100% sure about this or how it would work, but the problem seems easier to reproduce after killing and unkilling the drone a few times.

tomvand, 2018-07-18:
Tested on a second Bebop2 with the same results.

Managed to get gdbserver working on the Bebop2, but no conclusive results yet. For completeness:
```
Starting program: /home/tom/paparazzi/var/aircrafts/bebop2_opticflow/ap/ap.elf 
[New Thread 1359.1372]
[Switching to Thread 1359.1372]
0x000452f4 in image_to_grayscale ()
(gdb) bt
#0  0x000452f4 in image_to_grayscale ()
#1  0x00048758 in calc_fast9_lukas_kanade ()
#2  0x0004abe8 in opticflow_calc_frame ()
#3  0x0004afc8 in opticflow_module_calc ()
#4  0x00038094 in cv_run_device ()
#5  0x0004ba64 in video_thread_function ()
#6  0x000a8b04 in start_thread ()
#7  0x000cd168 in ?? ()
Backtrace stopped: frame did not save the PC

(gdb) thread apply all bt
[New Thread 1359.1366]
[New Thread 1359.1367]
[New Thread 1359.1368]
[New Thread 1359.1369]
[New Thread 1359.1370]
[New Thread 1359.1371]
[New Thread 1359.1373]

Thread 9 (Thread 1359.1373):
#0  0x000cbed4 in select ()
#1  0x000cd39c in __libc_enable_asynccancel ()
#2  0x00000000 in ?? ()

Thread 8 (Thread 1359.1371):
#0  0x000cb864 in poll ()
#1  0x000cd39c in __libc_enable_asynccancel ()
#2  0x43ea6d00 in ?? ()
Backtrace stopped: frame did not save the PC

Thread 7 (Thread 1359.1370):
#0  0x000cbed4 in select ()
#1  0x000cd39c in __libc_enable_asynccancel ()
#2  0x00000000 in ?? ()

Thread 6 (Thread 1359.1369):
#0  0x000ac248 in pthread_cond_wait ()
#1  0x00010c6c in i2c_thread ()
#2  0x000a8b04 in start_thread ()
#3  0x000cd168 in ?? ()
Backtrace stopped: frame did not save the PC

Thread 5 (Thread 1359.1368):
#0  0x000cbdfc in ioctl ()
#1  0x00010b6c in i2c_thread ()
#2  0x000a8b04 in start_thread ()
---Type <return> to continue, or q <return> to quit---
#3  0x000cd168 in ?? ()
Backtrace stopped: frame did not save the PC

Thread 4 (Thread 1359.1367):
#0  0x000ac248 in pthread_cond_wait ()
#1  0x00010c6c in i2c_thread ()
#2  0x000a8b04 in start_thread ()
#3  0x000cd168 in ?? ()
Backtrace stopped: frame did not save the PC

Thread 3 (Thread 1359.1366):
#0  0x000ad310 in read () at ../sysdeps/unix/syscall-template.S:84
#1  0x00011670 in sys_time_thread_main ()
#2  0x000a8b04 in start_thread ()
[... 703 lines cut ...]
#0  0x000cd700 in clock_gettime ()
#1  0x0001194c in get_sys_time_usec ()
#2  0x0005adec in imu_bebop_event ()
#3  0x0001045c in main () at firmwares/rotorcraft/main.c:70
Thread 3 (Thread 1368.1375):
#0  0x000ad310 in read () at ../sysdeps/unix/syscall-template.S:84
#1  0x00011670 in sys_time_thread_main ()
#2  0x000a8b04 in start_thread ()
#3  0x000cd168 in ?? ()
Backtrace stopped: frame did not save the PC

Thread 2 (Thread 1368.1382):
#0  0x000c8144 in strlen ()
#1  0x000e087c in vfprintf ()
---Type <return> to continue, or q <return> to quit---
#2  0x000cdd4c in __printf_chk ()
#3  0x03042a5a in ?? ()
Backtrace stopped: frame did not save the PC

Thread 1 (Thread 1368.1368):
#0  0x000cd700 in clock_gettime ()
#1  0x0001194c in get_sys_time_usec ()
#2  0x0005adec in imu_bebop_event ()
#3  0x0001045c in main () at firmwares/rotorcraft/main.c:70
```

OpenUAS, 2018-12-04:
Shall we retest after PR #2364  is honored? Would be good to have pinpointed cause before TU Course 2019 started though

OpenUAS, 2018-12-19:
OK @kirkscheper will fix #2364  after a "Stable" release, will get an interesting January before TU course 2019 starts...

kirkscheper, 2019-01-15:
This may be connected to #2361 #2378, a fix has been pushed for an error with the spi which is used by the sonar. The error would only surface under some circumstances which would explain why I couldn't recreate this. Can you test your airframe with master/rebase on master and let us know if you still have this issue?

tomvand, 2019-01-17:
Re-tested today with da3e040 and master (9ba84b8). The problem still occurs, I could get a segfault during takeoff with both commits. I could not replicate it above the textured mat, though.

tomvand, 2019-02-27:
Reopening this, the segfault still occurs in master with the bebop2_opticflow aircraft in the example conf.

ap.elf now consistently crashes in v4l2.c line 102. The `dev` struct appears incorrect, with an unreadable name and height of 0. The dev width changes between runs, so I wouldn't be surprised if this struct gets overwritten from somewhere else...

Crash log 1:
```
/data/ftp/internal_000/paparazzi # ./gdb ap.elf 
GNU gdb (GDB) 7.11
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-none-linux-gnueabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ap.elf...done.
(gdb) r
Starting program: /data/ftp/internal_000/paparazzi/ap.elf 
[New LWP 1424]
[New LWP 1425]
[New LWP 1426]
Current schedparam: policy 0, prio 0
Current schedparam: policy 0, prio 0
New schedparam: policy 1, prio 29
New schedparam: policy 1, prio 11
[New LWP 1427]
Current schedparam: policy 0, prio 0
New schedparam: policy 1, prio 10
[New LWP 1428]
Current schedparam: policy 0, prio 0
New schedparam: policy 1, prio 10
Current schedparam: policy 0, prio 0
New schedparam: policy 1, prio 10
[New LWP 1429]
Current schedparam: policy 0, prio 0
New schedparam: policy 1, prio 10
[New LWP 1430]
[New LWP 1431]
[video_thread-/dev/video0] Set nice level to 5.
[v4l2-capture] Select timeout on /dev/video0
ap.elf: pthread_mutex_lock.c:81: __pthread_mutex_lock: Assertion `mutex->__data.__owner == 0' failed.

Thread 9 "v4l2" received signal SIGSEGV, Segmentation fault.
[Switching to LWP 1431]
v4l2_capture_thread (data=0x199c58) at modules/computer_vision/lib/v4l/v4l2.c:102
102     modules/computer_vision/lib/v4l/v4l2.c: No such file or directory.
(gdb) bt
#0  v4l2_capture_thread (data=0x199c58) at modules/computer_vision/lib/v4l/v4l2.c:102
#1  0x000aec30 in start_thread (arg=0xb35ff300) at pthread_create.c:335
#2  0x000d3398 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) info locals
sr = <optimized out>
now_ts = 33662766
prev_idx = <optimized out>
dev = 0x199c58
buf = {index = 3, type = 1, bytesused = 115200, flags = 8197, field = 1, timestamp = {tv_sec = 1314, tv_usec = 237549}, timecode = {type = 0, flags = 0, frames = 0 '\000', seconds = 0 '\000', 
    minutes = 0 '\000', hours = 0 '\000', userbits = "\000\000\000"}, sequence = 2627, memory = 1, m = {offset = 356352, userptr = 356352, planes = 0x57000 <stateCalcVerticalWindspeed_f+64>, 
    fd = 356352}, length = 115200, reserved2 = 0, reserved = 0}
tv = {tv_sec = 1, tv_usec = 988806}
fds = {fds_bits = {1024, 0 <repeats 31 times>}}
__PRETTY_FUNCTION__ = "v4l2_capture_thread"
(gdb) print *dev
$1 = {name = 0x199c90 "\336", fd = 10, thread = 215, w = 26, h = 0, buffers_cnt = 5 '\005', buffers_deq_idx = 0 '\000', mutex = {__data = {__lock = 1, __count = 206, __owner = 72, __kind = 0, 
[... 59 lines cut ...]

Thread 9 "v4l2" received signal SIGSEGV, Segmentation fault.
[Switching to LWP 1329]
v4l2_capture_thread (data=0x195c58) at modules/computer_vision/lib/v4l/v4l2.c:102
102     modules/computer_vision/lib/v4l/v4l2.c: No such file or directory.
(gdb) bt
#0  v4l2_capture_thread (data=0x195c58) at modules/computer_vision/lib/v4l/v4l2.c:102
#1  0x000ac028 in start_thread (arg=0xb356e300) at pthread_create.c:335
#2  0x000d0748 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) info locals
sr = <optimized out>
now_ts = 227464515
prev_idx = <optimized out>
dev = 0x195c58
buf = {index = 0, type = 1, bytesused = 115200, flags = 8197, field = 1, timestamp = {tv_sec = 313, tv_usec = 39419}, timecode = {type = 0, flags = 0, frames = 0 '\000', seconds = 0 '\000', 
    minutes = 0 '\000', hours = 0 '\000', userbits = "\000\000\000"}, sequence = 19920, memory = 1, m = {offset = 0, userptr = 0, planes = 0x0, fd = 0}, length = 115200, reserved2 = 0, reserved = 0}
tv = {tv_sec = 1, tv_usec = 988810}
fds = {fds_bits = {1024, 0 <repeats 31 times>}}
__PRETTY_FUNCTION__ = "v4l2_capture_thread"
(gdb) print *dev
$1 = {name = 0x195c90 "\324", fd = 10, thread = 235, w = 22, h = 0, buffers_cnt = 5 '\005', buffers_deq_idx = 1 '\001', mutex = {__data = {__lock = 1, __count = 223, __owner = 28, __kind = 0, 
      __nusers = 0, {__spins = 216, __list = {__next = 0xd8}}}, __size = "\001\000\000\000\337\000\000\000\034", '\000' <repeats 11 times>, "\330\000\000", __align = 1}, buffers = 0x20}
(gdb) 
```

tomvand, 2019-02-27:
oops now actually reopening...

OpenUAS, 2019-03-27:
Closed with pull #2411

PR #2381 (2019-01-25): [cv] fix possible divide by zero
This should hopefully fix #2294
closed 2019-03-27 by OpenUAS