paparazzi/paparazzi #2298: Chimera doesn't compile

EwoudSmeur, 2018-07-19:
On master, the Chimera board doesn't compile any more, which seems to be because of this line:
https://github.com/paparazzi/paparazzi/blob/8dbff73844bbcbce7cc661daf011293a8f372a15/sw/airborne/boards/baro_board_ms5611_i2c.c#L31
Which was added by @kirkscheper in #1909 
Do we need this include here? Can it be removed? Or can we put it in an ifdef or something?

gautierhattenberger, 2018-07-19:
Solving this is part of https://github.com/paparazzi/paparazzi/pull/1910 that is still in discussion. I think you can extract the part that fix this specific issue and merge it to master already.

kirkscheper, 2018-07-19:
This is a duplicate of #2223. See that thread for some more details. I will try to test and merge #1910 by next Wednesday (pprz day).

closed 2018-07-19 by kirkscheper