paparazzi/paparazzi #2313: Vsupply resolution increase to 0.01V?

matejkarasek, 2018-08-09:
Currently the resolution of battery voltage (electrical.vsupply) is 0.1V, which is rather low if you fly on 1S battery. 
https://github.com/paparazzi/paparazzi/blob/018df2fb301c5960f8b857f03498e64d4cff6050/sw/airborne/subsystems/electrical.c#L128

How about increasing it to 0.01V? uint16 would still allow to store max 655V, so more than enough... Are there some reasons against this?

podhrmic, 2018-08-09:
You would have to change the units for each pprzlink message that uses `vsupply`: https://github.com/paparazzi/pprzlink/blob/master/message_definitions/v1.0/messages.xml#L803 (that is easy).

The bigger task is to change failsafe code that compares vsupply to some threshold to make it aware that we have a different resolution now (or convert from centivolts to decivolts as needed).

Indeed this wouldn't be backwards compatible (so we should make users aware of this change).

Otherwise I don't have any objections.

matejkarasek, 2018-08-09:
The failsafe code seems to be in the same file, with the hardcoded multiplication by 10
https://github.com/paparazzi/paparazzi/blob/018df2fb301c5960f8b857f03498e64d4cff6050/sw/airborne/subsystems/electrical.c#L184
So an easy fix there, just a different multiplier...

grep on 'electrical.vsupply' doesn't give that many hits, so even the rest of the code can be fixed easily

Of course, backward compatibility would be an issue (that's why I asked here...)

podhrmic, 2018-08-09:
I think it is a fine change for v5.13 release, just have to be transparent about it. @gautierhattenberger @fvantienen what do you think?

gautierhattenberger, 2018-08-09:
All the ground code should also be checked for that of course. I'm not really against this but I was hoping to improve the BAT/ENERGY messages in general. Their is no particular reason to keep both, the uint16 for energy in ENERGY msg can be a problem. I would be in favor of merging them and using float everywhere (and reflect that in the airborne code of course).
We could also find less intrusive solution like changing the airborne part for higher resolution (or float) and use the ENERGY message for display instead of BAT. This way, messages are unchanged.

fvantienen, 2018-08-11:
For me it is also fine to change it. But we could indeed better chamge it to float directly

OpenUAS, 2018-08-21:
Depend on messagesize .01 resolution anyone or alt_unit or...? But a Bat and Energy message cleanup welcomed. Make a proposed fix, PR the we can discuss it in this issue.

closed 2019-01-15 by kirkscheper

matejkarasek, 2019-01-15:
Thanks a lot @kirkscheper!