paparazzi/paparazzi #2317: Vectornav navigation offset in real flight

podhrmic, 2018-09-01:
When Vectornav is used as an INS, GCS shows an offset between the actual and desired flight path. This is likely due to some conversion errors between vectornav's float/double data and INS's internal fixed point representation.

[image]

OpenUAS, 2018-09-07:
Maybe related.. see screenshot, experience this (no not a 12 Beaufort hurricane sideslip ;)...was flinin nose in direction of circle. Had no time to investigate yet..
[image]

gautierhattenberger, 2018-09-07:
@OpenUAS I guess it is a simulation ? what was the wind estimate at that moment ?

@podhrmic I wouldn't expect something so regular if it was a quantization error. Maybe something wrong during init ? Did you try to compare lat/lon from vector nav and a standard GPS ?

podhrmic, 2018-09-09:
@gautierhattenberger good idea, that would be the next step.

podhrmic, 2018-09-20:
@gautierhattenberger I am trying to figure out if this is a problem on the GCS side, or the AP side.

What kind of coordinates does the GCS show? Lat/lon, ECEF or local?

That would help me narrow down where to look.

gautierhattenberger, 2018-09-20:
it is using lat/lon from the FLIGHT_PARAM message

gautierhattenberger, 2018-12-05:
@podhrmic did you found something for this issue ?

podhrmic, 2018-12-05:
Not yet, it is in the pipeline - waiting for better weather to flight test.

gautierhattenberger, 2018-12-05:
I think I know the problem. The INS is initialized with the rotorcraft LTP, which is a "true" LTP, while the fixedwing aircraft is abusing LTP as "local UTM". Since LTP and UTM axis are not exactly aligned, you will have this kind of small errors, probably bigger when you go further from the init point.
And you most likely have no errors when using this INS on rotorcraft.
The way to solve this I think, is to have a different initialization according to the firmware (see https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/ins/ins_float_invariant.c#L218). We should also check if the altitude should be above ellipsoid or above geoid when using FW aircraft.
@podhrmic if you don't think you'll have the time to handle this in the coming weeks, I'll remove the v5.14 milestone but keep the issue opened of course.

podhrmic, 2018-12-06:
So what do you think is a good solution? Should I add logic as if `INS_FINV_USE_UTM` is defined, or the other case? I would be adding logic into [ins_vectornav_init()](https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/ins/ins_vectornav.c#L163) function

gautierhattenberger, 2018-12-06:
there is no helper function to hide the initialization differences between firmware at the moment, so each INS should do that. This is something we should improve in future versions. So at the moment you should add the proper logic in your init function.

OpenUAS, 2020-12-17:
> @OpenUAS I guess it is a simulation ? what was the wind estimate at that moment ?
No it was a real flight. Issue is old now , so need to do a full retest with latest master and all fixes in there, can do that after PR #2629 is honored, preferably in period Dec2020/Jan2021

OpenUAS, 2020-12-17:
I'll also see to dig up an Verctornav and add it to AC, but also for that need normal testfly normal fixedwing with latest master first

gautierhattenberger, 2023-12-04:
Vectornav ins don't seem to be used anymore

closed 2023-12-04 by gautierhattenberger