paparazzi/paparazzi #2328: git branch cleaning

gautierhattenberger, 2018-09-14:
We have a bunch of branches in the paparazzi repository that are not pending pull request. I suggest that the authors of these branches decide if they are worth keeping or not:

  remotes/paparazzi/bitshift_cleanup
  remotes/paparazzi/blos
  remotes/paparazzi/campaign2011
  remotes/paparazzi/campaign2013
  remotes/paparazzi/coverity_scan
  remotes/paparazzi/cpp_support
  remotes/paparazzi/crypto_link
  remotes/paparazzi/cyberzoo_gazebo
  remotes/paparazzi/dshot_f4
  remotes/paparazzi/feature4_fixedwing_in_rotor
  remotes/paparazzi/gec_module_makefile
  remotes/paparazzi/lpe
  remotes/paparazzi/message_scheduling
  remotes/paparazzi/messages
  remotes/paparazzi/mission_simulation
  remotes/paparazzi/ocamlbuild
  remotes/paparazzi/rc_datalink
  remotes/paparazzi/revert-2010-fix_configs
  remotes/paparazzi/secure_link
  remotes/paparazzi/secure_link_tmp
  remotes/paparazzi/sonarqube_test
  remotes/paparazzi/state_utm

Some of them seems interesting, so they could be opened as PR

dewagter, 2018-09-14:
cleaned up mine

podhrmic, 2018-09-19:
cleaned up mine, see the related PR #2332 
also, keeping the `coverity_scan` branch for now - not sure if anyone (@kirkscheper ?) is using it.

gautierhattenberger, 2018-12-05:
closing this as most of the branches have been either removed or kept on purpose

closed 2018-12-05 by gautierhattenberger