paparazzi/paparazzi #2331: quat_of_axis_angle does not yield a normalized quaternion

joostmeulenbeld, 2018-09-18:
Both the int32_quat_of_axis_angle and float_quat_of_axis_angle don't return a normalized quaternion. This is due to the fact that

1. The axis is not normalized inside the function. The documentation does state that it uses a unit axis.
2. (only for the int32 variant) The PPRZ_ITRIG_COS and PPRZ_ITRIG_SIN do not give the angles that correspond with the quaternion fractions (i.e. PPRZ_ITRIG_COS(0) != QUAT1_BFP_OF_REAL(1))

Is this behavior as designed?
If it is, I think the function documentation should mention this behavior more clearly.
If not, setting the cosine and sine of the angle to the proper frac may do the job, although it's not completely clear to me how to work with a unit axis that is not either the x,y or z axis.

Anyway I could try to solve it/update the documentation if deemed required. Right guidance_hybrid_set_cmd_i in guidance_hybrid.c outputs a non-normalized quaternion. This is not a problem since the reference generator in stabilization_quat_int.c normalizes it. Normalizing q_yaw_sp in guidance_hybrid_set_cmd_i also works if the behavior  of int32_quat_of_axis_angle shouldn't change.

gautierhattenberger, 2018-09-18:
Although I didn't made those functions I wouldn't say this is desired behavior...
In the float case, the doc is not obvious, but I guess it is expecting a normalized vector to work properly. Forcing the normalization can also be a problem if a zero vector is passed by mistake.
In the integer case, it fails indeed since the resolution of quat is 15, for angle 12, for trigo 14 and unit vector is not specified at all...
I don't have a strong opinion except that the doc should be improved.

What about:
- keeping float with better doc for input/output
- make the int version at least return a quat with correct scale if the unit vector doesn't have fractional part:
```
void int32_quat_of_axis_angle(struct Int32Quat *q, struct Int32Vect3 *uv, int32_t angle)
{
  int32_t san2;
  PPRZ_ITRIG_SIN(san2, (angle / 2));
  int32_t can2;
  PPRZ_ITRIG_COS(can2, (angle / 2));
  q->qi = (can2 << INT32_QUAT_FRAC-INT32_TRIG_FRAC));
  q->qx = (san2 << INT32_QUAT_FRAC-INT32_TRIG_FRAC) * uv->x;
  q->qy = (san2 << INT32_QUAT_FRAC-INT32_TRIG_FRAC) * uv->y;
  q->qz = (san2 << INT32_QUAT_FRAC-INT32_TRIG_FRAC) * uv->z;
}
```
we could also pass an extra argument with the resolution of ov `uv` to make it correct in all case (`{1024, 0, 0}` is also a unit vector with a resolution of 10 :) )

gautierhattenberger, 2018-12-05:
@joostmeulenbeld do you have a comment on this ?

closed 2018-12-18 by gautierhattenberger

joostmeulenbeld, 2018-12-18:
@gautierhattenberger Sorry, I was preoccupied with finishing my thesis and the outback challenge.
I agree with your suggestions!

PR #2367 (2018-12-18): [math] fix quat_of_axis functions
Improve doc and choose a more reasonable output resolution of the integer version.

The only possible issue might be in the hybrid guidance:
- quat setpoint should now have the proper value, but since the "wrong" value was currently compensated later in the code it shouldn't change anything
- euler setpoint should be correct now, but it is unclear if it was even used with hybrid guidance (it shouldn't)
- hybrid guidance from master is currently not much used apparently and might need some extra cleaning and better integration anyway, better have this part corr [... cut]
commit bb35c16d (2018-12-18): [math] fix quat_of_axis functions (#2367)

improve doc and choose a more reasonable output resolution of the
integer version

fix #2331