paparazzi/paparazzi #2356: Generated autopilot: modules not included

podhrmic, 2018-11-25:
The modules included in the autopilot's xml file are not loaded:

```xml
<autopilot name="Fixed Wing Autopilot">

  <state_machine name="ap" freq="CONTROL_FREQUENCY" gcs_mode="true" settings_mode="true">

    <modules>
      <load name="some_module"/>
    </modules>

```

Not sure if it is intended behaviour - it would be cool to be able to autoload modules like this, but if the feature is not supported, we should remove modules from the `autopilot.dtd`

@gautierhattenberger ?

gautierhattenberger, 2018-11-26:
the modules from generated AP will be loaded at some point. It is just not implemented yet. Actually, I'm rewriting all the code generation part with a colleague, and it is almost done, so I'm not really willing to make the effort to fix this now, but rather finish the new version.

podhrmic, 2018-11-26:
That is OK, I am happy to wait for the new version. Thanks for the comment. I ll keep the issue open and close it after the new code is finished.

OpenUAS, 2018-12-04:
@gautierhattenberger :+1: for close to finish the new version

gautierhattenberger, 2018-12-05:
the new version of code generation won't be ready for next release, so I take this out of 5.14

gautierhattenberger, 2021-12-13:
done in v6.0

closed 2021-12-13 by gautierhattenberger