paparazzi/paparazzi #2378: Bebop airframes in master get 'Segmentation fault (core dumped)' with sonar

ls90911, 2019-01-14:
I pulled the latest master branch and tested Bebop airframes in tudelft/conf.xml.  I found that if I used the airframes which have sonar,for example Bebop_default,  then the system will crash with the information 'Segmentation fault (core dumped)'. 

I used printf() to debug and found that the codes ended up at line 144, 

spi_submit(&spi0, &sonar_bebop_spi_t);

Currently I am at the commit ' mark as v5.15 devel until next release'. 

Thanks!

gautierhattenberger, 2019-01-14:
do you mean that it is working if the sonar is not activated ?

ls90911, 2019-01-14:
> do you mean that it is working if the sonar is not activated ?

Yes. If I comment out <module name="sonar_bebop"/>, it works.

kirkscheper, 2019-01-15:
@ls90911 can you test if  #2379 fixes your issue?

ls90911, 2019-01-15:
> @ls90911 can you test if #2379 fixes your issue?

Yes, now it works! Thanks!

kirkscheper, 2019-01-15:
Fix merged to master

closed 2019-01-15 by kirkscheper