paparazzi/paparazzi #2404: Building paparazzi gives libpprzlink error

Christiaanvv, 2019-03-06:
When building Paparazzi the following issue occurs: 

fatal: reference is not a tree: 9e26b7a1aae5493d7fe77e48814aa428e5d73479
Unable to checkout '9e26b7a1aae5493d7fe77e48814aa428e5d73479' in submodule path 'sw/ext/pprzlink'
Makefile:46: recipe for target 'pprzlink.update' failed
make[1]: *** [pprzlink.update] Error 1
Makefile:144: recipe for target 'libpprzlink' failed
make: *** [libpprzlink] Error 2

Error occurs when building inside docker (image: flixr/pprz-dev), but also when building on local machine running Linux Mint 18.3.

kirkscheper, 2019-03-06:
Without any additional information as to which version of paparazzi you are trying to build and/or which branch you are using (I am going to assume it is not master as that points to 39735d5 and git branch -a --contains 9e26b7a1aae5493d7fe77e48814aa428e5d73479 says there is no commit with this SHA) I would say this is a user based git issue not a paparazzi issue. 

After a quick search through pprzlink commits, I couldn't find the one you are trying to checkout so it seems you are pointing to a different remote than paparazzi/pprzlink. To fix this you have to add the appropriate remote to the pprzlink submodule git.

As an example, if the remote to my commit is at my personal github I would do the following (in the command line pointing to somewhere inside the pprzlink folder) to add it to the git search:

git remote add kirk https://github.com/kirkscheper/pprzlink.git
git remote update -p

Obviously, you would have to add the actual remote above, not mine.

I hope this helps. If any of the assumptions I made are inaccurate or you have more information than you originally gave please feel free to add that info here.

EwoudSmeur, 2019-03-06:
Coincidentally, I just wanted to pull pprz and had exactly the same problem. Where do you find that pprz is pointing to 39735d5 ? For me it is pointing to that non-existing commit, even if I do a fresh clone. Should it point to pprzlink master?

dewagter, 2019-03-06:
did someone squash-merge a new message in pprzlink?
https://github.com/paparazzi/pprzlink/commits/master

master may now point to pprzlink 61192d861fba6dd56c79af0471b93377028bbba8

-Christophe

gautierhattenberger, 2019-03-06:
something went wrong after merging the PR for the joystick module, #2405 should fix this

PR #2405 (2019-03-06): fix ref to pprzlink
should fix #2404
closed 2019-03-06 by gautierhattenberger