paparazzi/paparazzi #2410: Vertical reference is wrong in guided mode

ls90911, 2019-03-26:
In my module, I keep call autopilot_guided_goto_ned(0.0,0.0,-1.5,0) with fixed frequency. However, the reference is not kept at -1.5. You can find the log here.
[image]

The problem might be that in **guidance_v_set_guided_z(float z)**, there are two lines

guidance_v_z_sum_err = 0; 
GuidanceVSetRef(stateGetPositionNed_i()->z, stateGetSpeedNed_i()->z, 0);

which update the reference based on current altitude.

OpenUAS, 2019-03-27:
@ls90911 Plz check if it is still an issue in latest master of today I could not reproduce it.

dewagter, 2019-03-27:
@OpenUAS bug is still there:

https://github.com/paparazzi/paparazzi/blob/50f574d4581d027f5edec207f106e9d81e4c2ded/sw/airborne/firmwares/rotorcraft/guidance/guidance_v.c#L565

This resets the reference to the current altitude every time you set the guided_z. So if you do that at 512Hz the drone is constantly at the perfect altitude and allowed to drift as much as it wants. If you **guided_v_set_guided_z** slower, then the slower, the better the altitude...

gautierhattenberger, 2019-04-08:
I can reproduce it today as well but didn't have time to try possible correction. Because the previous line is also a problem (setting integrator to zero). If I'm not wrong, we should just remove these two lines and also one in `set_guided_vz function`, then the ref should be normally updated from the `guided_run` function. But better try it before...

dewagter, 2019-04-08:
Yes I think they should only be in the on_enter function (certainly not called at 512Hz) and be removed in all set functions.

ls90911, 2019-04-09:
I have fixed it in my branch. Will send a PR soon.

PR #2418 (2019-05-17): [rotorcraft] don't reset vertical ref in guided mode
When calling the set_z or set_vz functions in guided mode, the reference
should not be called as it would make them ineffective when the control
rate is high

close #2410
commit 37cf050e (2019-05-17): [rotorcraft] don't reset vertical ref in guided mode (#2418)

When calling the set_z or set_vz functions in guided mode, the reference
should not be called as it would make them ineffective when the control
rate is high

close #2410
closed 2019-05-17 by gautierhattenberger