paparazzi/paparazzi #2412: direct input options for Lisa/M 2.0

colebunny, 2019-03-28:
Hello,

I am developing a novel class of hydrofoil sailboat and am using the paparazzi environment to control the hydrofoil which will be operating similarly to a fixed wing UAV. I am working with a Lisa/M 2.0 board with aspirin and need to be able to pass a 3 digit number to it that will be generated by an Arduino based steering mechanism the human pilot of the ship will be manipulating. 

My question is what port on the board is my best option for passing this input into. I haven't built the arduino part yet so the output from that which the Lisa will receive can be pretty much any format. 

Attached is a picture of my board with the connections that are already in use. I apologize the photo isn't that clear, I will update with a better photo next time I am with the hardware (traveling right now and this is only picture I have with me). 

Thank you so much! So grateful to have such a cool tool to work with. If anyone is particularly interested in the project I'm happy to discuss I just wanted to keep my question succinct. 

[Here'e the Lisa schematic 
](https://wiki.paparazziuav.org/wiki/Lisa/M_v2.0)
and here's my board 
[image]

podhrmic, 2019-03-28:
You could use any of the UART ports and send your number as one of the [PAYLOAD messages](https://github.com/paparazzi/pprzlink/blob/master/message_definitions/v1.0/messages.xml#L889)

colebunny, 2019-03-29:
I asked an EE friend via text for help and he gave me this response. Figured I'd add it incase somebody else comes across this- his solution is below

Cool. On the main board there are a few options open:
CAN
I2C
SPI
UART (maybe, cant see if this is actually open from the picture)

Of those, the only bus that is rated for at least 20m is CAN. A problem there is that Arduinos don’t speak CAN unless they have an extra board. https://www.sparkfun.com/products/13262

I2C and SPI *might* work but not recommended.

UART by itself also *might* work, but the more robust route is adding an RS-232, RS-422 or RS-485 transceiver to both the main board side and Arduino side.

I suggest RS-232.
There are so many options for boards to support this.
For main board side, one of the many options is: https://www.amazon.com/Hobbypower-Sp3232-Transceiver-Evaluation-Development/dp/B00I03VYOK
For Arduino side, there is a shield that is an easy solution: https://www.dfrobot.com/product-1030.html?gclid=Cj0KCQjw4fHkBRDcARIsACV58_GlricWh_fzFx2ywOVO-CRCqu2p0WMzTJ2dpzQI3eFyla8lKyTolDsaAga8EALw_wcB
 
Using RS-232 will add more boards to your product, but, if that is OK, it is the most reliable option and the one id go with without knowing more.

podhrmic, 2019-03-29:
@colebunny this looks to me more like an EE issue, rather than Paparazzi related issue. The specific details really depend on your application. From what you mentioned, UART/RS-232 seems like a solid choice.

colebunny, 2019-04-03:
@podhrmic Thank you for the help, missed that you had replied before I posted my solution.

closed 2019-04-03 by colebunny