paparazzi/paparazzi #2415: Fixedwing Safety Modes

dewagter, 2019-04-05:
- TOO_FAR_FROM_HOME with RC -> NO HOME (1.5*too far, throttle off works)
 - HOME mode throttle: 100% after boot and failsafe-throttle after GPS_ERROR

gautierhattenberger, 2019-04-08:
- RC modes are not affected by too_far_from home, only NAV: https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/fixedwing/autopilot_static.c#L128

- can it be a bug in the linux PWM driver ? how do you enter HOME mode at boot ?

gautierhattenberger, 2020-04-06:
- First point is not a bug at the moment, but should however be considered as the new European regulation is needing it -> make a dedicated Issue/PR for that
- If not more information on second point, I'll close this one

gautierhattenberger, 2020-04-30:
timeout

closed 2020-04-30 by gautierhattenberger