paparazzi/paparazzi #2417: initializing inputs + variables with a Lisa 2

colebunny, 2019-04-10:
Hello, 

I am using a Lisa M 2 with aspirin to control a hydrofoil sailboat that I am developing. (The foil flies and is controlled the same as a fixed wing aircraft so using all the open source stuff available here seemed like the way to go). 

Here’s my question…

Since the wing is underwater I have added a depth sensor which is connected to the Lisa via the GPIO port. Where in the my paparazzi files (airframe, settings, etc) is the appropriate place to initialize this hardware and store the variable that holds the depth value? I also have “desired heading”, and “desired depth” messages from the boat’s pilot coming in through the CAN/UART port which I assume would be initialized and stored in more or less the same way? 

The end goal is to have a very simple flight plan XML where I am using the heading function to control the foil’s depth and course with the values coming in from the pilot via CAN/UART. 

In a previous related question someone suggested I use the payload messages function in paparazzi to deal with these types of inputs but I couldn’t find good documentation on how to set up that feature in my airframe- just a list of messages that can be passed. 

Thank you so much for the help. I am in a little over my head but am willing to put in the work to make this thing fly. The resources and help I have gotten from the pprz community so far are impressive and inspiring. 

-Cole

gautierhattenberger, 2019-04-12:
Hello,
Normally, github bug tracker is not the best place for this kind of issues. We have a good old mailing list (http://lists.paparazziuav.org/) or the gitter channel (https://gitter.im/paparazzi/discuss) for that normally.

Here are some comments on your setup:
- you're saying that you want to read the depth sensor from a pin of the GPIO port, but this kind of port is digital so only able to give 0 or 1. What kind of sensor are you using ? Is it analog ? Then you'll need to use a pin from one of the two ANALOG connector.
- I can see a CAN/I2C connector, but not CAN/UART. Which port do you want to use exactly ?
- Communication can be done using the PPRZLINK library (https://github.com/paparazzi/pprzlink). There is a small documentation for  the Python API. What kind of device will send commands to the autopilot ?

In general, what you want to do is feasible, but not so trivial. You will need to write some sort of module (http://wiki.paparazziuav.org/wiki/Modules) to read your sensor and compute the proper command that will be applied to the actuators.

goonbunny, 2019-04-12:
Thank you so much for the response Gautier! From now on I'll post new questions on Gitter- didn't know that was the place for such things. I'll continue this post here though. 

The background on this project is actually that I am taking it over from a close friend who passed away. This project was his dream so I wanted to complete it for him. He assembled most of the hardware but was just starting on the software so I am trying to re-trace his steps and then carry the project to completion. 

1) Regarding the depth sensor I think I plugged it into the GPIO port and then forgot and assumed it had already been plugged in when I inherited the project. I can't find any info on the sensor I have so think I am going to upgrade to [this one](https://www.bluerobotics.com/store/sensors-sonars-cameras/sensors/bar02-sensor-r1-rp/). It should be compatible with the Lisa board and the paparazzi right? 

2) I meant CAN/I2C or UART. I think I am going to use the CAN though because this connecting cable may end up needing to be pretty long (20m) and I read CAN is more reliable at this distance. 

3) Thank you for showing me that library, that's really what I was looking for. I'll start reading up. The device that is sending commands to the Lisa is an Arduino with a tilt compensating compass and a couple buttons on it. Idea is that the pilot and point at the heading he wants, press a button, and the arduino will capture the heading and pass it to the Lisa. 

Thank you for your time and also you putting up with my beginner questions. I would really struggle without nice people like you acting as sign posts for what to study and am very grateful.

OpenUAS, 2019-06-08:
@goonbunny Can we close the issue?

goonbunny, 2019-06-10:
@OpenUAS  yes you can close it thank you. 

Also just a not I ended up solving my issue using a telemetry connection, a cable seemed more solid but field testing showed that the telemetry was actually more dependable and easier to manage.

closed 2019-06-10 by fvantienen