paparazzi/paparazzi #2420: YAW arming state machine incorrect

fvantienen, 2019-05-10:
The YAW arming state machine is incorrect. The following problem can occur:
- THRUST:0, YAW:0, ROLL:0, PITCH:0
- THRUST:0, YAW>9120, ROLL:0, PITCH:0
- Motors are turned on: `motors_on` and you can start flying
- If you keep either THRUST, YAW, ROLL or PITCH above 480 the whole time (which could be possible for THRUST, by making a diagonal move on the left stick) , the `in_flight` will be set to false continuously and the AP will never go to an armed state.

Suggested change is: changing `(autopilot_arming_check_valid(YAW_MUST_BE_CENTERED))` in the `STATUS_START_MOTORS` to only check for `YAW_STICK_CENTERED` no matter of all the other sticks.

gautierhattenberger, 2019-05-11:
I'm not really sure I get this. You mean that when pushing the yaw stick to the corner whit idle throttle you are still above threshold ?
I issue I can see is that if we only check for yaw stick centered, there is a way to enter `in_flight` state with full throttle straight.

fvantienen, 2019-05-13:
No you hit the yaw stick to the right with no throttle until the props turn on (motors_on). But then instead of keeping 0 throttle and turning the yaw back you keep the yaw and give a little bit of throttle. Then if you keep the throttle and move the yaw back you never go to in_flight and can still fly (without yaw control).

The problem is that the motors_on is triggerd at full yaw and then only when you have 0 yaw, throttle, pitch and roll the in_flight is triggered.

I see indeed that it is possible to enter in_flight with full throttle (but the plane or quadrotor is then actually full throttling already because the motors are on). So I dont really see this as a problem.

PR #2421 (2019-05-13): [airborne] Fix yaw arming
This should fix #2420 .
closed 2019-05-13 by gautierhattenberger