paparazzi/paparazzi #2426: Link Instability

Aeromus, 2019-05-14:
# The Issue
Noticeably Paparazzi's Link Program has a tendency to crash when presented with large messages. This issue was discovered whilst testing [etherlink](https://github.com/paparazzi/etherlink). 

## How to reproduce 

1. Program an aircraft and launch Paparazzi ground station so messages are being sent
2. Then use `ssh` to get into the gumstix co-pilot / Isaac (alternatively use `screen`)
3. Use `ivyprobe` with the `-b` flag to connect to your link
4. Use the following python script to generate a PAYLOAD_COMMAND message of n length and send it via `ivyprobe`
```python
def char_test(length):
    string="10 PAYLOAD_COMMAND 10 "
    for i in xrange(1, length):
        string += "%s,"%1
    string+="%s"%length
    print string
    print "Character length: %s"%len(string)
```
Noticeably messages up to a length of 247 will not crash the program, though a length of 248 will reliably crash the link, every time. Note that it is not the message length in characters that causes the issue but array length. This can be verified by changing the line `string += "%s,"%1` to `string += "%s,"%i`.

gautierhattenberger, 2019-05-14:
The structure of the message is 1 sync byte, 1 byte for total length', 4 bytes for message header, the actual payload, 2 CRC bytes. So the maximum length of the message including header and encapsulation is 255 = 247 + 8.
I agree that you should have a message or an exception when the array length is too big. Which API are using ? e.g. is it a ocaml, python, C or rust program that is crashing in your case ?

podhrmic, 2019-06-17:
@gautierhattenberger I just retested the issue. 

It seems to be related to the OCaml `link` program parsing what comes from the autopilot - or what comes over ivybus. 

I can reliably get this error:
```
Failure("PprzLink.values_of_payload, wrong argument: 08 00 01 72 85 00 00 00 08 00 ca 1a 9a 52 ca 2d e6 c5 b9 eb 9b 0f 08 00 45 00 00 72 81 9d 40 00 40 11 ad 89 c0 a8 45 01 c0 a8 45 02 d7 6c 1b 39 00 5e d8 e2 61 62 63 64 65 66 67 58 41 53 41 53 46 41 46 53 41 53 46 61 73 66 73 64 66 67 73 66 64 67 73 61 31 33 34 35 36 32 33 34 34 33 37 73 64 32 33 35 61 73 33 78 73 73 64 73 61 61 66 73 64 6b 6c 66 73 61 6c 67 68 73 61 6c 68 67 6b 68 66 62 67 76 6b 68 65 64 74 0a ")
```

anytime I send a message over 100 bytes total.

The interesting thing is that in the above message, anything at 11th byte and further matches what I can see in wireshark (i.e. it is a valid ethernet packet as expected). However, the first 10 bytes are extra, and I suspect it is something like a header of pprzlink packet etc.

Hence I suspect the parser gets a failiure on incoming pprzlink packets.

Do these bytes look familiar to you?
```
08 00 01 72 85 00 00 00 08 00
```

It is for DL_COMMAND message

OpenUAS, 2019-06-18:
Could a non updated custom conf/messages,xml be the cause. and before a build not done a make clean  to reflect the changes?

OpenUAS, 2019-06-18:
@Aeromus could you attach var/include/pprzlink/telemetry/COMMANDS.h and var/include/pprzlink/datalink/COMMANDS_RAW.h  to this issue?

gautierhattenberger, 2019-06-18:
Your message seems to be:
- sender id: 08
- receiver: 00 (GCS)
- class: 1 (telemetry)
- msg: 72 (114: PAYLOAD)
- number of elements in payload array: 85 (133 !?)
- the bytes
- last two should be checksum

Is this matching what you send or want to send ?

podhrmic, 2019-06-18:
@gautierhattenberger thanks for the quick response! Yes that matches - I am sending 133 bytes of payload, and the header is correct as well. It looks like the parser strips the STX byte and the checksum before displaying the error message, so those are not included. Is it possible that the checksum failed or would that be a different error?

I checked the internal buffer size at the autopilot, and it is set to 254 so it should not be an issue.

This is where the autopilot send the payload messages: https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/modules/mission/copilot_common.c#L184

And I am using a direct serial connection for testing, so that shouldn't be an issue either.

@OpenUAS Here is the PAYLOAD.h - at first glance it looks sane to me:
```C
/**
 * Sends a PAYLOAD message (API V2.0 version)
 *
 * @param msg the pprzlink_msg structure for this message
 * @param nb_values,_values 
 */
static inline void pprzlink_msg_v2_send_PAYLOAD(struct pprzlink_msg * msg, uint8_t nb_values, uint8_t *_values) {
#if PPRZLINK_ENABLE_FD
  long _FD = 0; /* can be an address, an index, a file descriptor, ... */
#endif
  const uint8_t size = msg->trans->size_of(msg, /* msg header overhead */4+1+nb_values*1);
  if (msg->trans->check_available_space(msg, _FD_ADDR, size)) {
    msg->trans->count_bytes(msg, size);
    msg->trans->start_message(msg, _FD, /* msg header overhead */4+1+nb_values*1);
    msg->trans->put_bytes(msg, _FD, DL_TYPE_UINT8, DL_FORMAT_SCALAR, &(msg->sender_id), 1);
    msg->trans->put_named_byte(msg, _FD, DL_TYPE_UINT8, DL_FORMAT_SCALAR, msg->receiver_id, NULL);
    uint8_t comp_class = (msg->component_id & 0x0F) << 4 | (1 & 0x0F);
    msg->trans->put_named_byte(msg, _FD, DL_TYPE_UINT8, DL_FORMAT_SCALAR, comp_class, NULL);
    msg->trans->put_named_byte(msg, _FD, DL_TYPE_UINT8, DL_FORMAT_SCALAR, DL_PAYLOAD, "PAYLOAD");
    msg->trans->put_bytes(msg, _FD, DL_TYPE_ARRAY_LENGTH, DL_FORMAT_SCALAR, (void *) &nb_values, 1);
    msg->trans->put_bytes(msg, _FD, DL_TYPE_UINT8, DL_FORMAT_ARRAY, (void *) _values, nb_values*1);
    msg->trans->end_message(msg, _FD);
  } else
        msg->trans->overrun(msg);
}
```

podhrmic, 2019-06-18:
I can see the failure happens here: https://github.com/paparazzi/pprzlink/blob/61192d861fba6dd56c79af0471b93377028bbba8/lib/v2.0/ocaml/pprzLink.ml#L670

but  I am not proficient enough with OCaml to see where in the `try` block it failed. Maybe the payload wasn't parsed correctly?

gautierhattenberger, 2019-06-18:
We have to find which function is raising the Invalid_Argument, and the only candidate I could find so far is the Array.init from `value_of_bin` function, which can hardly happen...
You can eventually try this to print the full error message (not sure it actually works)
```
        Invalid_argument e ->
              failwith (sprintf "PprzLink.values_of_payload, wrong argument: %s (%s)" (DebugPL.xprint buffer) e)
```

podhrmic, 2019-06-18:
@Aeromus can you pls try out what @gautierhattenberger suggested?

Aeromus, 2019-06-24:
@podhrmic @gautierhattenberger , I added / changed the line to ```Invalid_argument e ->
              failwith (sprintf "PprzLink.values_of_payload, wrong argument: %s (%s)" (DebugPL.xprint buffer) e)```

gautierhattenberger, 2019-06-24:
did you recompile correctly the library ? any error or warning ?

Aeromus, 2019-06-24:
ran a `make clean && make` on paparazzi, there were no notable errors or warnings

gautierhattenberger, 2019-06-24:
When you said that their is no changes, do you mean that even the error message is unchanged (so you don't see the `()` nor a test between the parenthesis ?
I guess you have changed the file `sw/ext/pprzlink/lib/v2.0/ocaml/pprzLink.ml` ? When typing `make` it is possible that the changes are discarded unless you have committed the modification of the subsystem pprzlink. Eventually, you can only recompile the pprzlink lib with `make pprzlink_protocol`.

Aeromus, 2019-06-24:
The error messages currently look like this: 
`
Failure("PprzLink.values_of_payload, wrong argument: 0b 72 a5 00 00 00 08 00 01 00 5e 00 00 fb 4a 5b 4e d5 3b db 08 00 45 00 00 92 1e 70 40 00 ff 11 76 44 c0 a8 45 02 e0 00 00 fb 14 e9 14 e9 00 7e 72 65 00 00 00 00 00 07 00 00 00 00 00 00 04 5f 66 74 70 04 5f 74 63 70 05 6c 6f 63 61 6c 00 00 0c 00 01 04 5f 6e 66 73 c0 11 00 0c 00 01 0b 5f 61 66 70 6f 76 65 72 74 63 70 c0 11 00 0c 00 01 04 5f 73 6d 62 c0 11 00 0c 00 01 09 5f 73 66 74 70 2d 73 73 68 c0 11 00 0c 00 01 08 5f 77 65 62 64 61 76 73 c0 11 00 0c 00 01 07 5f 77 65 62 64 61 76 c0 11 00 0c 00 01 ")
Failure("PprzLink.values_of_payload, wrong argument: 0b 72 a5 00 00 00 08 00 01 00 5e 00 00 fb 4a 5b 4e d5 3b db 08 00 45 00 00 92 1e 90 40 00 ff 11 76 24 c0 a8 45 02 e0 00 00 fb 14 e9 14 e9 00 7e 72 65 00 00 00 00 00 07 00 00 00 00 00 00 04 5f 66 74 70 04 5f 74 63 70 05 6c 6f 63 61 6c 00 00 0c 00 01 04 5f 6e 66 73 c0 11 00 0c 00 01 0b 5f 61 66 70 6f 76 65 72 74 63 70 c0 11 00 0c 00 01 04 5f 73 6d 62 c0 11 00 0c 00 01 09 5f 73 66 74 70 2d 73 73 68 c0 11 00 0c 00 01 08 5f 77 65 62 64 61 76 73 c0 11 00 0c 00 01 07 5f 77 65 62 64 61 76 c0 11 00 0c 00 01 ")
Failure("PprzLink.values_of_payload, wrong argument: 0b 72 a5 00 00 00 08 00 01 00 5e 00 00 fb 4a 5b 4e d5 3b db 08 00 45 00 00 92 1f ce 40 00 ff 11 74 e6 c0 a8 45 02 e0 00 00 fb 14 e9 14 e9 00 7e 72 65 00 00 00 00 00 07 00 00 00 00 00 00 04 5f 66 74 70 04 5f 74 63 70 05 6c 6f 63 61 6c 00 00 0c 00 01 04 5f 6e 66 73 c0 11 00 0c 00 01 0b 5f 61 66 70 6f 76 65 72 74 63 70 c0 11 00 0c 00 01 04 5f 73 6d 62 c0 11 00 0c 00 01 09 5f 73 66 74 70 2d 73 73 68 c0 11 00 0c 00 01 08 5f 77 65 62 64 61 76 73 c0 11 00 0c 00 01 07 5f 77 65 62 64 61 76 c0 11 00 0c 00 01 ")
Failure("PprzLink.values_of_payload, wrong argument: 0b 72 a5 00 00 00 08 00 01 00 5e 00 00 fb 4a 5b 4e d5 3b db 08 00 45 00 00 92 20 71 40 00 ff 11 74 43 c0 a8 45 02 e0 00 00 fb 14 e9 14 e9 00 7e 72 65 00 00 00 00 00 07 00 00 00 00 00 00 04 5f 66 74 70 04 5f 74 63 70 05 6c 6f 63 61 6c 00 00 0c 00 01 04 5f 6e 66 73 c0 11 00 0c 00 01 0b 5f 61 66 70 6f 76 65 72 74 63 70 c0 11 00 0c 00 01 04 5f 73 6d 62 c0 11 00 0c 00 01 09 5f 73 66 74 70 2d 73 73 68 c0 11 00 0c 00 01 08 5f 77 65 62 64 61 76 73 c0 11 00 0c 00 01 07 5f 77 65 62 64 61 76 c0 11 00 0c 00 01 ")`

And yes I changed the file in `sw/ext/pprzlink/lib/v2.0/pprzLink.ml` and it appears the change is still there so I do not believe it was dropped when running make.

gautierhattenberger, 2019-06-24:
If you changed with `(%s)` at the end of the string, it should at least appear at the end of the error message. Eventually, you should replace the original message completely to be sure that the correct code is being called.

Aeromus, 2019-06-24:
Upon further inspection it appears that the change is being omitted when I remake paparazzi. Is there a way to force it to make the file with the new changes?

gautierhattenberger, 2019-06-24:
- option 1: after making the changes, `git add sw/ext/chibios` and `git commit`, then you can do the usual `make`
- option 2: after making the changes, `make pprzlink_protocol` and `make -C sw/ground_segment/tmtc` (not that you need to properly define PAPARAZZI_HOME and PAPARAZZI_SRC environment variables for that)

podhrmic, 2019-06-26:
@gautierhattenberger I have an answer - the error apparently comes from `Array.init`, see below:
```
Failure("TEST: PprzLink.values_of_payload, wrong argument: 01 00 01 72 f5 80 00 00 08 00 b6 31 b8 eb 92 67 6e 2d 7e b0 74 b1 08 00 45 00 02 40 29 5d 40 00 40 06 04 07 c0 a8 45 02 c0 a8 45 01 00 16 b8 d0 f9 cb a8 2b aa d7 94 77 80 10 00 52 58 18 00 00 01 01 08 0a 02 89 26 78 4c 24 ce 44 00 00 04 34 06 14 a4 08 cb ac 2d 44 29 6f fd 6b c2 5b 05 be 5e 8d 00 00 01 02 63 75 72 76 65 32 35 35 31 39 2d 73 68 61 32 35 36 2c 63 75 72 76 65 32 35 35 31 39 2d 73 68 61 32 35 36 40 6c 69 62 73 73 68 2e 6f 72 67 2c 65 63 64 68 2d 73 68 61 32 2d 6e 69 73 74 70 32 35 36 2c 65 63 64 68 2d 73 68 61 32 2d 6e 69 73 74 70 33 38 34 2c 65 63 64 68 2d 73 68 61 32 2d 6e 69 73 74 70 35 32 31 2c 64 69 66 66 69 65 2d 68 65 6c 6c 6d 61 6e 2d 67 72 6f 75 70 2d 65 78 63 68 61 6e 67 65 2d 73 68 61 32 35 36 2c 64 69 66 66 69 65 2d  (Array.init)")
```

gautierhattenberger, 2019-06-26:
This was my first guess (https://github.com/paparazzi/paparazzi/issues/2426#issuecomment-503319929) so this means that the init size is not valid (file:///usr/share/doc/ocaml-doc/ocaml.html/libref/Array.html).
I think the issue is coming from the fact that the size of the array is read as a int8 (https://github.com/paparazzi/pprzlink/blob/master/lib/v2.0/ocaml/pprzLink.ml#L439) so any array larger than 127 bytes will trigger this issue. You can eventually check this hypothesis.
If it is the actual problem, it means that a variable length array in a message should be smaller than 127 bytes, or we need to extract an uint8 instead of an int8.
Note that their is currently no protection against the max size of a message (size encoded on a single byte).

podhrmic, 2019-06-26:
That was totally it!
Changing `c_int8_of_indexed_bytes` to:
```
value
c_int8_of_indexed_bytes(value s, value index)
{
  CAMLparam2 (s, index);
  uint8_t *x = (uint8_t*)(String_val(s) + Int_val(index));
  CAMLreturn (Val_int(*x));
}
```
makes it work with longer arrays!

If you are OK with this fix, I will submit a PR for that. FYI @Aeromus

gautierhattenberger, 2019-06-27:
It is a bad idea to change the return type of this function, it may have a lot of side effect.
I think the correct way is to change https://github.com/paparazzi/pprzlink/blob/master/lib/v2.0/ocaml/pprzLink.ml#L439 by:
```
 let n = Char.code buffer.[index] in
```

gautierhattenberger, 2020-01-06:
fixed with https://github.com/paparazzi/pprzlink/pull/102

closed 2020-01-06 by gautierhattenberger