paparazzi/paparazzi #2428: Use the fixed wing paparazzi autopilot in jsbsim ?

alexandrequ, 2019-05-15:
Hello, 
Is it possible to use the fixed wing paparazzi autopilot with JSBSim in standalone mode ? 
Thanks for your help ! 
 
Alex

gautierhattenberger, 2019-05-15:
Not sure what you exactly want to do, but here is an example of fixed-wing airframe with a target `nps` for doing simulation with JSBSIM: https://github.com/paparazzi/paparazzi/blob/master/conf/airframes/examples/microjet_lisa_m.xml#L13

As a side not, please prefer the github bug tracker for actual software bug reports. You can use the mailing list http://lists.paparazziuav.org/ or the gitter chat room https://gitter.im/paparazzi/discuss for this type general question. Thanks

OpenUAS, 2019-06-18:
@alexandrequ I assume you got it to work, since no reaction on the issue here for a month. therefore closing your issue. If not satisfied plz contact the developers via Paparazzi channels mentions by @gautierhattenberger

closed 2019-06-18 by OpenUAS