paparazzi/paparazzi #2429: flightplan XML files result in ParseError: invalid token

danielwillemsen, 2019-05-16:
In flight_plan configs, often the invalid XML character "&" is used, this causes errors when parsing these files using XML-parsers such as the python **xml.etree.ElementTree** module. This character occurs often in the exception part of the flight_plan.

Example: **/tudelft/course2018_orangeavoid_cyberzo.xml** contains:
`    <exception cond="(!InsideCyberZoo(GetPosX(), GetPosY()) &&
      !(IndexOfBlock('Holding point') > nav_block) &&
      !(nav_block >= IndexOfBlock('Land here')) &&
      (autopilot_in_flight() == true) )" deroute="Land here"/>`

When running:
`xml.etree.ElementTree.parse("conf/flight_plans/tudelft/course2018_orangeavoid_cyberzoo.xml").getroot()`

Results in error:
` ParseError: not well-formed (invalid token): line 51, column 43`

This also still happens with other invalid characters such as "<", as described in #154

dewagter, 2019-05-16:
- Most XML editors also complain about the paparazzi flightplan XML.
 - many flightplans inverse the comparisons to avoid the error of <
 - there are defines to avoid other xml problems

Would it not be an option to replace all illegal XML with stubs like [And] and [IsSmallerThan] which the generator turns into C-code?

I think this (5 > alt) code is harder to check than:
(alt [IsSmallerThan] 5)

gautierhattenberger, 2019-05-20:
All fields are not processed the same way, but it might still be possible in most cases

gautierhattenberger, 2019-05-20:
something like this ? https://github.com/gautierhattenberger/paparazzi/commit/d37ec754f85ed550dad8dca5f9d6adb457e8db56

fvantienen, 2019-05-21:
That looks much better

dewagter, 2019-05-27:
I like it too, and so does my XML editor.

PR #2434 (2019-05-31): [ocaml] add special keyword to avoid conflict with XML invalid characters
close #2429
closed 2019-05-31 by gautierhattenberger