paparazzi/paparazzi #2432: how is the RGB JPEG images as RTP payload data calculated

driesj, 2019-05-23:
when I look at the data scan of an jpeg file with 64x48 pixels using [online hex editor](https://www.onlinehexeditor.com/#), I get 936 values. But when I look at your code at [rtp.c](https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/modules/computer_vision/lib/encoding/rtp.c) I see that your data is only 56 values long.
I know that 0xff and 0xd9 are EOI indicators, so that would only make place for 54 values.

So I want to know how you get this RTP payload from a 64x48 jpeg image and why it is only 54 values long?

fvantienen, 2019-05-24:
The image headers of the jpeg file is removed, because this will be the same for all images. The only data transmitted are the image pixels.

driesj, 2019-05-24:
and how are these image pixels received? is that the raw data from the file excluding the headers?

driesj, 2019-05-24:
for example if I get the raw data of an 48x32 pixel image I get 654 values. So, that means that the data must be compressed in a way, but i don't know what way.

fvantienen, 2019-05-24:
it is compressed with JPEG, and you can define how hard you want to compress

driesj, 2019-05-24:
To be more specific this data is found between Start of scan (SOS) and end of image (EOI) with the EOI included?

OpenUAS, 2019-06-08:
@driesj  If this issues is resolved plz let us know, so we can close the issue. Better to use the mailing list or ask via Gitter. Making an issue in the sourcecode repop issue tracker not ideal since basically not an issue but a generic question.

closed 2019-06-08 by driesj