paparazzi/paparazzi #2439: no-sized-deallocation breaks build on defauld Ubuntu arm compiler

OpenUAS, 2019-06-18:
On default compilers of Ubuntu e.g. for arm gcc v4.9 there is no -fno-sized-deallocation command option.

As such wil break building with:

arm-none-eabi-g++: error: unrecognized command line option '-fno-sized-deallocation'

To update a compiler is more often than not an option with many machines and luckily also not needed.

The reason for this behavior comes from the treatment of files with .cpp. pure C should never be linked to this but is

In GCC there is an option -x ()-x c, -x c++, -x objective-c, -x assembler-with-cpp)
This to specify the source language. This has nothing to do with standards conformance or extensions; it merely selects which base syntax to expect.
If you give none of these options, cpp will deduce the language from the extension of the source file: .c, .cc, .m, or .S.

If cpp(C Pre Processor) does not recognize the extension, it will treat the file as C, *which is not what we want*

So therefore the extension .cpp is renamed to .cc and the C Pre Processor handles now correctly as it should :)

gautierhattenberger, 2019-06-18:
GCC 4.9 is known to have issues, and it is strongly recommended to use series 5 or above.
The file c++.cpp should only be needed when using C++ code I guess. The extension `.cpp` cause that this file is compiled with `g++` as it should, but also the the final program is linked with `g++`, which is not strictly needed.
I would be in favor of a solution where `c++.cpp` is not included to the sources if not needed.

fvantienen, 2019-06-18:
There could be a simple `ifeq` check to check if the CPP sources are not empty and then only include this file.

OpenUAS, 2019-06-18:
While I fully agree the solution where c++.cpp is not included to the sources is much better, for me time to craft a better solution is just not there ATM

So for now how to progress, can this PR be honoured as a temp fix for now?

gautierhattenberger, 2019-06-18:
This should work I guess:
```
-$(TARGET).srcs += c++.cpp pprz_syscalls.c
+$(TARGET).srcs += pprz_syscalls.c
+ifneq (,$(filter %.cpp, $($(TARGET).srcs)))
+$(TARGET).srcs += c++.cpp
+endif
```

OpenUAS, 2019-10-30:
It does, can compile, fix look fine. It works on a dozen other PC's also. Will make a new PR then close this issue.

OpenUAS, 2019-10-31:
Well 3Rd party EKF lib was dragged in the airframe using EKF2 based AHRS on that lib e.g. Splash3 we will get e.g. airspeed_fusion.cpp so same issue. Refactoring while possible, not realistic and compatible with original, so the "official" solution will not work (Except with symlink hackery which we want to avoid) so a better solution must be found.. ongoing.

OpenUAS, 2019-10-31:
Will not fix, no time, installed arm-none-eabi-g++ 7.3 and will do so on dozen other PCs and focus on other issues. Isn't the developers chaos lovely ;)

closed 2019-10-31 by OpenUAS