paparazzi/paparazzi #2444: NavDetectGround() makes flair deroute to holding point and kills motors.

danielwillemsen, 2019-06-21:
In many rotorcraft flight plans, the flare block is structured the following way:
`    <block name="flare">
      <exception cond="NavDetectGround()" deroute="Holding point"/>
      <exception cond="!nav_is_in_flight()" deroute="landed"/>
      <call_once fun="NavStartDetectGround()"/>
      <stay climb="nav_descend_vspeed" vmode="climb" wp="TD"/>
    </block>`

It seems that the flair is often derouted to holding point, this causes the airframe to fall out of the sky. This has been noticed when flying the course2018_orangeavoid_cyberzoo Bebop aircraft. However, also when using different (custom) airframes, based on a default bebop with optitrack with different flight plans (a modified version of rotorcraft_basic) this happens. It has only been tested with bebop airframes.

For course2018_orangeavoid_cyberzoo:
- It  happens very often, except when the takeoff is performed very smoothly by limiting the amount of throttle used.
- It does not happen if THRESHOLD_GROUND_DETECT is set tot a very high value (= 100 m/s^2).

#530 is possibly related.

gautierhattenberger, 2019-06-21:
The detection is done based on accelerometer threshold. So if you have a strong noise or if the limit is set too low or two high, it will not work correctly (either cut to soon or too late).

So the first thing is to adjust the threshold according to your aircraft (https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/autopilot_firmware.c#L77) or find a better criteria/filter (https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/autopilot_firmware.c#L212).

You can also try to deroute to a block that continuously reduce the throttle rather than cutting it.

fvantienen, 2019-06-21:
I would love to have a feature which detects a free-fall and then quickly starts the motors back again. So I think a module for landing rotorcraft would be ideal which could also include this safety feature.

My thoughts about how this module should work:
- Start to descend with a constant speed and does a ground detection
- Whenever ground is detected set thrust to a minimum (but keep props turning)
- If free-fall is detected go to step 1 and quickly spin-up the propellors
- If the props are ~3 seconds at a minimum and no free-fall then turn the props off

OpenUAS, 2019-10-31:
A freefall detector is somewhere in Wilco work I recall, we tested throwing hand launch and monitored for top of freefall