paparazzi/paparazzi #2451: Bebop 1 front camera too dark

tomvand, 2019-07-08:
After merging the latest master into the tu delft course code, I've noticed that the Bebop 1 front camera image is a lot darker than before. I suspect this has to do with #2364 . I have not found the cause yet as that PR made a lot of changes to the front camera. The default values are way too dark for flights inside the cyberzoo, I've tried increasing the color gains or exposure value but ran into some other issues (low framerates or even loss of video). I will play around with this a bit more, but any hints on what has changed or how to fix this would be most welcome.

tomvand, 2019-07-09:
For reference, the following values seem to work well in the cyberzoo:
```
    <define name="GAIN_GREEN1" value="10.0"/>
    <define name="GAIN_GREEN2" value="10.0"/>
    <define name="GAIN_BLUE" value="12.5"/>
    <define name="GAIN_RED" value="9.0"/>
    <define name="TARGET_EXPOSURE" value="30" />
```
The color gains are roughly a factor 2 larger than before.

dewagter, 2023-02-10:
Working gains are now in master + release.

closed 2023-02-10 by dewagter