paparazzi/paparazzi #2454: Inconsistent handling of define elements inside airframe xml

tomvand, 2019-07-12:
While trying to find a way to set up an arbitrary number of color filters from an airframe file, I noticed the following inconsistencies in the way the `<define ...>` elements are handled:

- The `type` attribute does not work correctly for defines inside the target element. It is not possible to specify an array type (e.g. `type="int[]"`). This works correctly within the section element, however.
- The define value can not contain a function call when placed inside the target element (e.g. `<define name="COLOR_OBJECT_FILTERS" value="{filter(0, 12, 13, 14);}"/>`. This causes a syntax error, "(" unexpected. Putting a similar define inside the section element works correctly.

gautierhattenberger, 2019-07-17:
This is because the defines in sections are generating `#define XXX YYY` in a header file, while the defines in firmware and target sections are generating compilation flags in makefiles `-DXXX=YYY` (where `=YYY` is optional BTW).
So you can't mix them in complex cases like array or function definition.
Except changing the names (which is troublesome), improving the error messages (just really annoying) or make a decent documentation (a what ?), I don't have a good solution for you, sorry.

tomvand, 2019-07-18:
Oh right, I was not aware of that but that explains a lot! I found the corresponding lines in Makefile.ac and played around a bit more but did not get anything to work, although it should theoretically be possible (https://stackoverflow.com/questions/31857559/gcc-define-function-like-macros-using-d-argument/31857844). But even if it worked it would have looked very hacky :P.
I agree that there is not really a quick fix for this, but at least it is documented here now which is a start.

I'll close this issue as it's not necessarily a defect and difficult to fix for something that might be hardly ever used in practice.

closed 2019-07-18 by tomvand