paparazzi/paparazzi #2459: How does the Pixhawk autopilot perform hardware-in-the-loop simulation on paparazzi?

Shaofeng757, 2019-09-29:
I have a pixhawk autopilot. The existing data only describes how to perform hardware-in-the-loop simulation of LIsa MX. How does the Pixhawk autopilot perform hardware-in-the-loop simulation on paparazzi?

podhrmic, 2019-09-30:
From [Configuration](https://wiki.paparazziuav.org/wiki/HITL#Configuration) section:

> HITL can currently run on any that has:
> * Serial port for Vectornav INS input (provides position and orientation data, including GPS coordinates)
> * Serial port for additional high-speed telemetry output (so not your regular 57600 telemetry)
> * other serial/io for regular telemetry, RC input etc.

So depending on which version of Pixhawk you have and whether it has enough serial ports, it can run in HITL mode. The mechanics are the same, it might be as easy as just changing the autopilot board to from LisaMX Pixhawk (in the HITL example config file).

Shaofeng757, 2019-10-01:
Thank you for your reply. The autopilot version I am using is Pixhawk 2.4.8. Can this Pixhawk version support HITL?

podhrmic, 2019-10-01:
I think you can use the serial port that goes to GPS for the additional high-speed telemetry output, so most likely yes.

OpenUAS, 2019-10-31:
@Shaofeng757 Thanks for your interest in Paparazzi HITL. You have more chance of answers on http://lists.paparazziuav.org/ and on https://gitter.im/paparazzi/discuss  after all your question, not defect, was paosted in an issue tracker not a forum ;)

closed 2019-10-31 by OpenUAS