paparazzi/paparazzi #2468: Bebop video streaming really slow and not usable

TommasoBendinelli, 2019-10-24:
I made a cleaned installation of Paparazzi. I have a parrot bebop2 and I tried to use video_rtp_stream for the video streaming. I used RTPviewer for visualization and ffmpeg.
There are many missing packets, I do not understand why. Furthermore, even if I stop all the processes (SERVER, GCS and data link) the bebop keeps me sending data. I do not think this a nice behaviour

noether, 2019-10-24:
I tried the same before summer with a bebop2, and I had the same bad performance.

I do not think whether there is a way to do it efficiently. At least, it is not in the wiki I believe.

TommasoBendinelli, 2019-10-24:
I think the issue is in the transmission of the packets. I tried the ros sdk for the parrot bebop and there I could stream in real time HD from the bebop, hence something is not optimised properly here.

gautierhattenberger, 2019-10-24:
It is clearly not optimized. Parrot is doing hardware compression with DSP or GPU to stream h26x video slow while we are doing mjpeg/rtp using software compression on the VPU. Mostly because we don't really have access to the GPU part. The main problem with RTP stream is that if you miss a single packet, you loose the complete frame. This is not the case with the format used by Parrot.
We don't pay too attention to that as our is not to make a nice video stream (original one is way better anyway), but to process images on-board for autonomous navigation.
If you really need to use video stream, you need to provide more information about your setup and goals. Maybe the there are some tweaking that can be done to avoid losing to many packets. It is also possible that the RTP encoding is not good or that at least it could be improved.

TommasoBendinelli, 2019-10-24:
Thank you for the quick answer.  My setup is a bebop 2 connected to ubuntu. My goal would be having access both IMUs sensors and the camera video for doing real time visual inertia odometry. Through the normal sdk Imu is not accessible directly and the states of the drone are just accessible at a limited rate of 5 kHz

TommasoBendinelli, 2019-10-24:
Why is not the GPU not accessible?

gautierhattenberger, 2019-10-24:
it is, but we have no real support from Parrot for that, so it is quite complicated. I don't know much about GPU myself, I think the guys from TUDelft have tried to use it, not sure what was the result.
Maybe your best shot is to analyze why you are losing packets, if it is due to collisions on the wifi link for instance. You can change the telemetry file of your drone to reduce the frequency of messages you don't need.
Eventually implement something more robust than RTP...

OpenUAS, 2019-10-31:
@TommasoBendinelli  It is possible to have a smooth output, just do not use the MJPEG but use hardware assisted H264 output, take a look at this course

https://github.com/tudelft/paparazzi/tree/mavlabCourse2019

And use that branch for now to test your stream If we ever find time to redo the entrypoint from the  H encoding lib, will try to merge to master.  (Parrot FW 4.0 and higher updated the encoding lib of Hardware assisted H264 and we found no time to make PPRZ compatible again...) 

BTW Take a look here...
https://github.com/tudelft/paparazzi/tree/merge_wilco_into_master

"hence something is not optimised properly here." could be...using hex-rays IDA was already not trivial in the past but could help again...

All in all it is very fixable if one dives into it...

OpenUAS, 2019-12-05:
@TommasoBendinelli any luck in trying to use the HW accelerated branch?

TommasoBendinelli, 2019-12-07:
Unfortunately I have not anymore the drone as I had to pass to a next project, hence I didn't go further.

OpenUAS, 2020-03-11:
OK thanks for reporting not the less

closed 2020-03-11 by OpenUAS