paparazzi/paparazzi #2483: Architecture proposal for a new GCS

Fabien-B, 2020-01-24:
There was a discussion about a new GCS one and a half year ago (https://github.com/paparazzi/paparazzi/projects/2). Lets continue this discussion!
I have a proposal for the architecture of a new GCS : https://github.com/Fabien-B/new_pprz_GCS/
Can you please react on this proposal ?

I choose Qt and C++ for its robustness and its popularity.

I mention you because you were on the original discussion: @podhrmic @gautierhattenberger @fvantienen @esden

noether, 2020-01-24:
Very happy about seeing things moving!

Fabien, I am not a sw developer, so probably I do not know whether the following makes sense.

I remember writing a GCS (in fact my first GCS xD) in C++ where the widgets/functionalities were independent threads following the subscriber/publisher. In that way, if a widget failed, the rest were continuing working, and my GCS was still up.

Does such a feature make sense here? In the end, I remember to have 10/15 default widgets and were quite robust. On the other hand, "heavy" widgets did not block other "light ones" if they were executed sequentially. However, I do not know whether that makes sense anymore. In the past, I was executing the GCS in an ASUS EEE Notebook and made sense xD.

Fabien-B, 2020-01-24:
The Qt documentation states :
>  The Qt GUI must run in [the main thread]. All widgets and several related classes, for example QPixmap, don't work in secondary threads.

So I think it doesn't apply really well here, but it will be up to each Widget to spawn their own threads if needed.

podhrmic, 2020-01-24:
I am happy to see things moving as well, thanks for the ping @Fabien-B :-)

Lets see if we are at the same starting point. In my understanding, the new GCS is needed first because the old Ocaml + GtK GCS is not being supported on new Ubuntu distros. Second thing are desired new features (better maps, planning etc) that people mentioned in the previous discussion.

I can see Qt being a nice replacement for the old GCS, mostly preserving the functionality, possibly adding some new functions.

But I still think there is value in exploring if we can adopt some of the other existing GCS (in reality, now there is only QGroundControl), through some plugin or so.

OpenUAS, 2020-01-24:
Yeah,@Fabien-B Great, nice and good that you set up the framework first. Will join en give response. BTW if you look at the VESC PC client project for good ideas ( https://github.com/vedderb/vesc_tool ). Benjamin made it so pull code, it compiled at first try... not the usual hassle as with some QT projects. If we still have the original flexibility see eg. for a frivolous experiment 
[image]  . Will see to get resources that help

Fabien-B, 2020-01-28:
@OpenUAS I am looking at the VESC client, interesting, although I think its more "static" than what we want for the GCS. I will continue looking to catch some "good practices" for Qt.

@podhrmic Yes, the main reason for me is also that OCaml + Gtk goes toward its end of support, and its hard to find people willing to code in OCaml (I do it only under torture), so C++/Qt is the more future proof choice for the support and its popularity, while still keeping "some" safety compared to python (it has to be compiled).
I try to see what we can do with QGroundControl but I don't know if all paparazzi concepts will be easily translated to the QGroundControl environment. Also, I tried looking for how to use local custom map but I couldn't find anything convincing, and I think it was one of the major feature requested for a new GCS.

What I proposed so far is just a simple framework allowing to easily integrate new widgets, but the main work will take place in those widgets, especially the map. For example, the "papgets" will be a functionality of the map.
A widget could have its own configuration file. We just have to add a "conf" attribute to the widget node in the layout file.
And we can also imagine that some widgets can have multiple versions to fit special needs.

noether, 2020-01-30:
> The Qt documentation states :
> 
> > The Qt GUI must run in [the main thread]. All widgets and several related classes, for example QPixmap, don't work in secondary threads.
> 
> So I think it doesn't apply really well here, but it will be up to each Widget to spawn their own threads if needed.

Yep. I remember we wrote that GCS with GTK2, and it was the same. We were constraint by other requirements, though, so we were "forced" to have multithread and we worked it around. I do not think this is a requirement anymore here :P. I am happy with the decision and that you started some code.

In the past, I wanted to add some extra visualizations to the GCS (the GVF and some multiagent information). If we have something working in C++, definitively I will be able to add them.

Thanks Fabien.

Fabien-B, 2020-02-12:
Hi everyone, update on this topic:
I've been working on a tiled map in pure C++/Qt those past weeks, and I think the map is now pretty ok.
It can display multiple layers of standard tiled maps of any size. If you want to give it a try, here is the repo: https://github.com/Fabien-B/GcsAgain
It builds with CMake.  Just change the path to store tiles, which is still hard coded in the file src/widgets/pprzmap.cpp, line 19.
I hope it will work well, please tell me if you tested, and especially if you found bugs.
I began to handle items on top of the map, right click on it to place a circle attached to a position on the map.
Now I will do something better to choose layers from the UI.

And why I haven't took some already working lib in javascript ? I don't really remember, but I had a reason (I think). At least, now you can store your own tiles on your drive...

[image]

closed 2020-12-03 by Fabien-B