paparazzi/paparazzi #2531: Adding fabs and explicit type conversion

joezie, 2020-05-24:
I found two bugs in navigation-related source codes. Details are listed below:

* nav_smooth.c (the earliest version with the bug: tree e2ccf1b6a317d1fb1a9b3c6d1f5376b5f94ab912 in release v5.17_devel)
https://github.com/paparazzi/paparazzi/blob/e2ccf1b6a317d1fb1a9b3c6d1f5376b5f94ab912/sw/airborne/modules/nav/nav_smooth.c#L157-L170
_delta_ is impacted by the values of _wind_east_, _wind_north_, and _airspeed_. In some cases, the computed _delta_ can be a negative value, and then we perform _sqrt()_ on it, which result to NaN error. To reproduce this error,  you can manually set _stateGetHorizontalWindspeed_f()->y_ and _stateGetHorizontalWindspeed_f()->x_ to -10.0 and 10.0 respectively at the following function call and run flight simulation.
https://github.com/paparazzi/paparazzi/blob/e2ccf1b6a317d1fb1a9b3c6d1f5376b5f94ab912/sw/airborne/modules/nav/nav_smooth.c#L189
To fix this bug, a possible solution is to perform _fabs()_ on _delta_ before performing _sqrt()_. 
`ground_speeds[i] = scal + sqrt(fabs(delta))/2.; `
But please note that the computed result might not be the ideal value.

* navigation.c (the earliest version with the bug: tree dd0c6cc1faa93b4132627334650b3243102b7073 in release v5.17_devel)
https://github.com/paparazzi/paparazzi/blob/dd0c6cc1faa93b4132627334650b3243102b7073/sw/airborne/firmwares/rotorcraft/navigation.c#L274
Here, _wp_diff_ and _pos_diff_ are two-dimensional variables representing the horizontal distance
between two waypoints _wp_start_ and _wp_end_ and between the UAV and _wp_start_ respectively.
The types of their members x and y are both signed integers. However, since _nav_leg_length_
is an unsigned integer, the result computed within the parentheses must be implicitly converted
to unsigned integer before divided by _nav_leg_length_. When the UAV is approaching the waypoint
_wp_start_, if the result computed within the parentheses happens to be a negative value whose most significant bit is set to 1, it would be interpreted as a large value after conversion to unsigned
integer. After being divided by _nav_leg_length_, the most significant bit becomes 0, and therefore,
the final result is converted back to a very large signed integer and assigned to _nav_leg_progress_.
https://github.com/paparazzi/paparazzi/blob/dd0c6cc1faa93b4132627334650b3243102b7073/sw/airborne/firmwares/rotorcraft/navigation.c#L278
Later, _nav_leg_progress_ is bounded within the range [0, prog_2], and consequently be trimmed to the upper bound _prog_2_. This value is not reasonable because _nav_leg_progress_ reflects the navigation progress, so it should not have shown a completed progress before straight navigation begins. To reproduce this error, you can set the start and end waypoint in a relatively short distance (so that _nav_leg_length_ is relatively small), and have the UAV fly towards the start waypoint from a direction that could make the computation result of _(pos_diff.x * wp_diff.x + pos_diff.y * wp_diff.y)_ to be a negative value. 
To fix this bug, we can add explicit type conversion to _nav_leg_length_:
`nav_leg_progress = (pos_diff.x * wp_diff.x + pos_diff.y * wp_diff.y) / (int32_t)nav_leg_length;`
Even the computation result within the parentheses is a negative value and the computed _nav_leg_progress_ is consequently a negative value, it would then be bounded within the range [0, prog_2], and consequently be trimmed to the lower bound 0. This would be a much reasonable value to show a starting progress of navigation.

noether, 2020-05-25:
Hi @joezie . Thanks a lot for spotting these two issues.

The second fix looks alright to me.

As you pointed out, I am also not sure about the first fix. I think the first issue should be fixed by checking the value of the airspeed (in case a sensor does not give an admissible value) to guarantee that delta is positive for all the alphas. On the other hand, I did not write this part, and I would need some time to see the whole picture/algorithm.

What do you think @gautierhattenberger ?

gautierhattenberger, 2020-05-25:
I also agree for the second point.

The first algo actually makes no sense if your airspeed is lower that the wind speed, since you would never reach the point.
So, I agree that checking for consistent inputs would be better, but using abs is not the right way I think.

noether, 2020-05-25:
Yeah, it seems that we can check if airspeed => windspeed, so we have always delta => 0 (no idea now whether there are corner cases because of the operations with floats).

The question now would be to think about how to handle the "exception" if airspeed < windspeed.

gautierhattenberger, 2020-05-25:
Since this navigation function is not able to handle this case anyway, I would just "quit" (return false) and let the flight plan handle the rest.

joezie, 2020-05-25:
I agree with the idea to return false on "exception".

closed 2020-05-26 by gautierhattenberger