paparazzi/paparazzi #2537: heading drift in simulation

EwoudSmeur, 2020-05-28:
In the nps (JSBSIM) simulator, I observe a slow heading drift of ~-20 degrees per 5 minutes for my cyfoam airframe. The scaled mag sensor values don't change as I just let the vehicle sit there on virtual ground.

The mag is rotated in the airframe, so I have in the mag configuration:
    <module name="mag" type="hmc58xx">
      <configure name="MAG_HMC58XX_I2C_DEV" value="i2c2"/>
      <define name="MODULE_HMC58XX_UPDATE_AHRS"/>

      <define name="HMC58XX_CHAN_X" value="1"/>
      <define name="HMC58XX_CHAN_Y" value="0"/>
      <define name="HMC58XX_CHAN_Z" value="2"/>

      <define name="HMC58XX_CHAN_X_SIGN" value="-"/>
      <define name="HMC58XX_CHAN_Y_SIGN" value="+"/>
      <define name="HMC58XX_CHAN_Z_SIGN" value="+"/>

      
      <define name="MPU9250_MAG_PRESCALER" value="32768"/>

      <define name="HMC58XX_MAG_TO_IMU_PHI" value="RadOfDeg\(6.5\)"/>
      <define name="HMC58XX_MAG_TO_IMU_THETA" value="RadOfDeg\(28.5\)"/>
      <define name="HMC58XX_MAG_TO_IMU_PSI" value="RadOfDeg\(181.0\)"/>
    </module>

And 

    <define name="BODY_TO_IMU_PHI"   value="0." unit="deg"/>
    <define name="BODY_TO_IMU_THETA" value="94." unit="deg"/>
    <define name="BODY_TO_IMU_PSI"   value="0." unit="deg"/>

I imagined that mag rotation to be the cause, but commenting that out as well as the sign stuff still does not work. Maybe someone has an idea? Otherwise I can investigate further...

EwoudSmeur, 2020-05-28:
Example: 
[image]

gautierhattenberger, 2020-05-29:
that is annoying for sure
do you have the same for a standard frame like a bebop ?
what do you have for the raw imu data ?

EwoudSmeur, 2020-07-10:
I thought I had tested this, but apparently I didn't report on this. I'm trying to test it now, but I have issues running the simulator on master. My Bebop_actuators A/C doesn't take off in the sim, and my bebop2_22 A/C doesn't align... Could it have anything to do with #2545? I can also not get my MODULES_FREQUENCY to be different from 60 for my tudelft/cyfoam.xml airframe, unless I start editing the Makefile.ac

gautierhattenberger, 2020-07-12:
Yes, this is a possibility that the module frequency have affected in some cases. Could you tell me the exact conf you are using?
Not taking off is more surprising...

gautierhattenberger, 2020-07-13:
Ewoud, someday you'll receive the 'best bug finder award'. The frequency issue is a bit more serious due to the new organization of the code generation. So I'll have to do something that I wanted to do for some time: stop computing the static prescalers during code generation, but at compile or runtime.
Dammit, I was going to have 2 days off...

EwoudSmeur, 2020-07-13:
Haha I always find a way to keep you busy :wink: 
If you still need to know, I'm using the tudelft conf.