paparazzi/paparazzi #2538: Running Joystick with ac GCS fails

EwoudSmeur, 2020-06-08:
When running the fc_rotor.py script, it should be able to take joystick inputs. Looking at #2246, I think it should be done by using the setting -ac GCS when running the joystick tool. However, that results in:

```
RUN '/home/ewoud/paparazzi/sw/ground_segment/joystick/input2ivy  -ac GCS frsky_tarranis_X9D_plus.xml -v'
Joystick ID (option -id): 129
Joystick SDL device index (option -d): 0
Fatal error: exception Failure("Couldn't parse message DL_SETTING (IndexOfSetting: unknown var 'autopilot.kill_throttle')")
```
Not sure why the autopilot_kill_throttle is coming from. Any ideas?

gautierhattenberger, 2020-06-11:
The problem is coming from the joystick XML file. To use the fc_rotor tool, you need to configure your stick to send the `JOYSTICK` message like in https://github.com/paparazzi/paparazzi/blob/master/conf/joystick/T1600M_ground.xml

closed 2020-06-11 by gautierhattenberger

EwoudSmeur, 2020-06-14:
Ah I see, thanks Gautier!