paparazzi/paparazzi #2541: CHIBIOS - Lisa-MX 2.1 - USE_ACTUATOR_7AND8

dewagter, 2020-06-30:
How to activate the PWM driver 4 in CHIBIOS?

 - e.g. aggieair_ark_quad_lisa_mx.xml
 - add ```<define name="USE_SERVOS_7AND8"/>``` in the actuators

Results in an error that ```PWMD4``` driver is not defined (yet). Any tips?

```
arch/chibios/subsystems/actuators/actuators_pwm_arch.c: In function 'actuators_pwm_arch_init':
arch/chibios/subsystems/actuators/actuators_pwm_arch.c:156:13: error: 'PWMD4' undeclared (first use in this function); did you mean 'PWMD5'?
   pwmStart(&PWMD4, &pwmcfg4);
             ^~~~~
             PWMD5
```

fvantienen, 2020-06-30:
`STM32_PWM_USE_TIM4` in `mcuconf.h` needs to be enabled.

dewagter, 2020-06-30:
Should I 

```
#ifdef USE_SERVOS_7AND8
#define STM32_PWM_USE_TIM4                  TRUE
#else
#define STM32_PWM_USE_TIM4                  FALSE
#endif
```

or can the otherwise unused TIM4 always be turned on

```#define STM32_PWM_USE_TIM4                  TRUE```

closed 2020-06-30 by dewagter