paparazzi/paparazzi #2546: DShot jitters

EwoudSmeur, 2020-07-02:
@fvantienen helped me to set up dshot on the Chimera autopilot (STM F7) with ChibiOS, but I'm having some issues. I can start and stop the motors, they spin and react to throttle changes. However, they jitter when changing the command. Sometimes I can find a stick position in which they stay at a constant RPM.

I have recorded the messages and they seem correct, at least most of the time. However, I just saw that sometimes there can be an extra bit, so 17 instead of 16. I can imagine that this could cause jitter. Is this a known issue @gautierhattenberger ? I recorded some data with a logic analyzer which I can share if needed. Thanks for any help!

EwoudSmeur, 2020-07-02:
[image]

The first 11 bits of the sent messages

gautierhattenberger, 2020-07-03:
I will forawrd the issue to my colleague who wrote the low level driver for this. If you have the recording of the logic analyzer that would help a lot.

EwoudSmeur, 2020-07-03:
Thanks Gautier!
This is the logic analyzer data and a matlab script for making the figure above:
https://filesender.surf.nl/?s=download&token=87cdfc90-049e-4494-863c-87e36948a2a0

This is how often 17 bits (34/2) are sent instead of 16:
[image]

gautierhattenberger, 2020-07-03:
Which timer are you using for the DSHOT ? If possible, link to the board file where this is configured. Thanks

EwoudSmeur, 2020-07-03:
Thanks for your response, we were using timer 3:
https://github.com/EwoudSmeur/paparazzi/commit/d3782b2a4f78a6fd0ba73bd716e8ab8491e277fb

alex31, 2020-07-03:
I am a bit away from Logic Analyser and i cannot verify that the candidate that i have found is actually the problem you have met, but i have a candidate : the dma buffer is not declared in a static memory area that is non cached, so the driver will work on F4 but will meet cache coherency problem on F7.

could you try this : in file
sw/airborne/arch/chibios/modules/actuators/actuators_dshot_arch.c.
replace all 
DSHOTDriver DSHOTDx;
by
IN_DMA_SECTION_NOINIT(DSHOTDriver DSHOTDx);

the macro IN_DMA_SECTION is declared in sw/airborne/arch/chibios/mcu_periph/ram_arch.h

gautierhattenberger, 2020-07-03:
you can grab the patch here: https://github.com/enacuavlab/paparazzi/commit/a2e368d953816c01b334254fd568533586fb26f4

EwoudSmeur, 2020-07-03:
Thanks @alex31 and @gautierhattenberger ! I can try on Monday morning :smiley:

EwoudSmeur, 2020-07-06:
I just tried it and I still got jitter (maybe it is a little bit less? hard to say). I checked and the number of bits is still 17 in 13% of the messages.
The logic analyzer recording is here: https://filesender.surf.nl/?s=download&token=eead5a5b-64aa-48e7-951d-0c15a56cf140
And this is what the number in the first 11 bits looks like over time with me varying the command:
[image]

EwoudSmeur, 2020-07-06:
If I look at that graph, if the number is below 1024, the extra bit divides the number by two and if it is above, it is divided by 2 and 1024 is added. I'd say that maybe the first bit is sometimes sent twice?

alex31, 2020-07-06:
Thanks for reporting, I will have a Logic Analyser on thursday, to make unit testing outside of pprz.

alex31, 2020-07-06:
to be able to reproduce your configuration, how many motor(s) are driven ? which timer(s) do you use ?

alex31, 2020-07-06:
If you are in a hurry, and you can spend some time on trials, you can disable dma mburst : 
file esc_dshot.c, line 150, replace 
`    .mburst = DSHOT_DMA_BUFFER_SIZE % (timerWidthInBytes * 4) ? 0 : 4,`
by
`    .mburst =  0,`

else, i will work on it on thursday

EwoudSmeur, 2020-07-06:
Thanks @alex31 , I am using timer 3 (https://github.com/EwoudSmeur/paparazzi/commit/d3782b2a4f78a6fd0ba73bd716e8ab8491e277fb#diff-d5d5e3e51d3206aa899667103a9f45e7R57) and I have two motors.
I am not really a hurry, and today I probably don't have time, but if I get the chance I'll give it a try!

EwoudSmeur, 2020-07-07:
@alex31 I had a quick test, but no luck with the mburst unfortunately.

alex31, 2020-07-07:
thanks for the test, the logic analyser is on his way and i will get it thursday evening

closed 2020-08-03 by EwoudSmeur