paparazzi/paparazzi #2563: Setting changes in flight log not traceable

EwoudSmeur, 2020-08-03:
When changing settings during a flight, this is logged in the GCS log with the 'SETTING' message. This message contains the setting ID. The .log file contains all setting definitions, but the setting ID is not in the .log file, and cannot be found by straightforward counting of dl_setting tags. The generated settings.h file contains the setting ID's, but this one is overwritten all the time. Is it possible to find the settings order somehow, or to save it in the .log file?

gautierhattenberger, 2020-08-04:
The settings ID should be in the same order than in the xml log file. What do you mean by not straightforward ? Is it because all settings are not at the same level of XML node ?
I'm not completely sure where should I add this in the build process but I will think about it.

EwoudSmeur, 2020-08-05:
What I mean is that in this case, I know which settings I changed and in the data file I find settings ID 22 and 23 are changed. If I look in the corresponding log file, at the corresponding aircraft section, I can see all the settings like this:
```
      <settings>
        <settings>
          <dl_settings>
            <dl_settings NAME="FollowMe">
              <dl_setting VAR="follow_me_distance" UNIT="m" STEP="0.1" SHORTNAME="distance" MIN="0." MAX="50."/>
              <dl_setting VAR="follow_me_height" UNIT="m" STEP="0.1" SHORTNAME="height" MIN="0." MAX="100."/>
              <dl_setting VAR="follow_me_heading" STEP="1.0" SHORTNAME="heading" MIN="0." MAX="360."/>
              <dl_setting VAR="follow_me_min_speed" STEP="0.1" SHORTNAME="min_speed" MIN="0." MAX="100."/>
              <dl_setting VAR="follow_me_filt" STEP="0.01" SHORTNAME="filt_val" MIN="0." MAX="1.0"/>
              <dl_setting VAR="follow_me_diag_speed" STEP="0.01" SHORTNAME="diag_speed" MIN="0." MAX="1.0"/>
              <dl_setting VAR="follow_me_gps_delay" UNIT="ms" STEP="1.0" SHORTNAME="gps_delay" MIN="0." MAX="10000.0"/>
              <dl_setting VAR="follow_me_datalink_delay" UNIT="ms" STEP="1.0" SHORTNAME="dlink_delay" MIN="0." MAX="20000.0"/>
              <dl_setting VAR="follow_me_advance_ms" UNIT="ms" STEP="1.0" SHORTNAME="advance" MIN="0." MAX="10000.0"/>
              <dl_setting VAR="follow_me_min_dist" UNIT="m" STEP="0.1" SHORTNAME="min_dist" MIN="0." MAX="100.0"/>
              <dl_setting VAR="follow_me_min_height" UNIT="m" STEP="0.1" SHORTNAME="min_height" MIN="0." MAX="100.0"/>
            </dl_settings>
          </dl_settings>
        </settings>
      </settings>
      <settings>
        <settings>
          <dl_settings>
            <dl_settings NAME="motor_flap_slider">
              <dl_setting VAR="pitch_slider" STEP="0.1" SHORTNAME="pitch" MIN="0.0" MAX="1.0"/>
              <dl_setting VAR="yaw_slider" STEP="0.1" SHORTNAME="yaw" MIN="0.0" MAX="1.0"/>
            </dl_settings>
```
The pitch_slider and yaw_slider are settings 12 and 13 in this case if I count from the top. In the generated settings.h I do see the same numbers:
```
case 0: telemetry_mode_Main = _value; break;\
    case 1: mag_lis3mdl_mag_lis3mdl_report_status = _value; break;\
    case 2: calibration_send_imu_mag_current_status = _value; break;\
    case 3: ekf2_ins_ekf2_update_status = _value; break;\
    case 4: gps_ubx_ucenter_gps_ubx_ucenter_periodic_status = _value; break;\
    case 5: gps_ublox_gps_ubx_periodic_check_status = _value; break;\
    case 6: flight_recorder_flight_recorder_periodic_status = _value; break;\
    case 7: autopilot_mode_auto2 = _value; break;\
    case 8: autopilot_KillThrottle( _value ); _value = autopilot.kill_throttle; break;\
    case 9: autopilot_SetPowerSwitch( _value ); _value = autopilot.power_switch; break;\
    case 10: autopilot_SetModeHandler( _value ); _value = autopilot.mode; break;\
    case 11: follow_me_distance = _value; break;\
    case 12: follow_me_height = _value; break;\
    case 13: follow_me_heading = _value; break;\
    case 14: follow_me_min_speed = _value; break;\
    case 15: follow_me_filt = _value; break;\
    case 16: follow_me_diag_speed = _value; break;\
    case 17: follow_me_gps_delay = _value; break;\
    case 18: follow_me_datalink_delay = _value; break;\
    case 19: follow_me_advance_ms = _value; break;\
    case 20: follow_me_min_dist = _value; break;\
    case 21: follow_me_min_height = _value; break;\
    case 22: pitch_slider = _value; break;\
    case 23: yaw_slider = _value; break;\
```
so I am wondering if there is a way to know which setting is which using the log file and without the settings.h file.

gautierhattenberger, 2020-08-05:
Which version / branch are you currently using ? It seems that the system settings are missing. Actually, it seems that it is even worse now with the new code generation. The system settings are still missing, but also module settings apparently...
I'll check this one....

EwoudSmeur, 2020-08-06:
For these tests we are unfortunately on a very old branch (1.5 years). Sorry I hadn't checked the behavior on master. Let me know if I can help solve the issue in any way!

gautierhattenberger, 2020-08-06:
Then, if you want to have the correct index, you'll need to count the missing one, which are the telemetry processes, the modules periodic function that are not "locked", the flight plan variables, etc. So, not something easy...
I don't know if I will fix that in older versions, but I'll fix it for the latest master, which is even worse at the moment.

EwoudSmeur, 2020-08-07:
That is great @gautierhattenberger , we should update our branch anyway, don't bother with the old versions!

gautierhattenberger, 2020-08-18:
@EwoudSmeur, I hate you, you are not allowed to find shitty bugs or corner case for at least a month !

see #2569

EwoudSmeur, 2020-08-19:
Haha I love you too @gautierhattenberger :wink:

Thanks a lot for fixing, I will give it a test soon!

PR #2569 (2020-08-26): Fix settings in log
[log] add the latest generated settings if available to log file
    
    This way, the exact order of the settings used during the flight might
    be available.
    The only false case would be to start a server, rebuild and aircraft
    with different parameters or target, then flash and start the plane. In
    this case, there is no guarantee that the information is correct.
    However, this is the best we can do, since we can't know when starting
    the server with which target the aircraft will be used.
    
    close #2563

Also fix the log replay
commit 6ad7fa93 (2020-08-26): Fix settings in log (#2569)

* [log] add the latest generated settings if available to log file

This way, the exact order of the settings used during the flight might
be available.
The only false case would be to start a server, rebuild and aircraft
with different parameters or target, then flash and start the plane. In
this case, there is no guarantee that the information is correct.
However, this is the best we can do, since we can't know when starting
the server with which target the aircraft will be used.

close #2563

* [fix] use correct FP dumping tool for replay
closed 2020-08-26 by gautierhattenberger