paparazzi/paparazzi #2570: ChibiOS log creates extra files

EwoudSmeur, 2020-08-19:
When using the chibiOS sd logger, next to the log also some extra files are created with jibberish names (e.g. '8┌'$'\a''F·'$'\006''≈╦.ÑHQ'). 
My log file is usually also there, but the file browser won't even open the PPRZ folder. In the terminal I get:
```
ls: cannot access 'Npl═■⌠╓b.à'$'\031\035': Input/output error
ls: cannot access 'NReh^±┐√.#.'$'\020': Input/output error
ls: cannot access ''$'\032''*]Sεl√C.q'$'\004''─': Input/output error
ls: reading directory '.': Input/output error
```
And then I can still see my log files and download them.

gautierhattenberger, 2020-09-16:
It can happen once your SD have been corrupted. Did you try to run fsck an clean the dirty bit ?

EwoudSmeur, 2020-09-21:
Thanks for your suggestion. In the end I had only tried formatting it on Linux, formatting on Windows did the trick. Thanks for the tip, I'll try fsck next time!

closed 2020-09-21 by EwoudSmeur