paparazzi/paparazzi #2576: compile use terminal linux

mohammad78741, 2020-09-06:
hi i trying compile use this command for compile:make -C /home/mohammad/paparazzi -f Makefile.ac AIRCRAFT=bebop2 ap.compile 
but i have this error:
Makefile:45: *** Architecture not set, maybe you forgot to add the target? e.g. <target name="tunnel" board="twog_1.0"/>.  Stop
please help me thanks.

EwoudSmeur, 2020-09-07:
You can compile on the command line, but it is easier to call: ./select_conf.py, and select the user configuration (set of airframes) that you want to use. Then you get the paparazzi center user interface, where you can select an airframe, compile it, and upload/simulate.

EwoudSmeur, 2020-09-07:
Oh and before that it is good to do a top level make:
~/paparazzi$ make clean && make

OpenUAS, 2020-11-14:
@mohammad78741 feel free to reopen if you feel the issue is not handled like you expected and you have more information of the tips given by @EwoudSmeur worked good enough for you.

closed 2020-11-14 by OpenUAS