paparazzi/paparazzi #2599: pprzros command signals limited to 8 bit resolution due to input2ivy

ddou210, 2020-10-09:
Hi everybody,

Recently I tried to send the attitude and thrust commands to the drone via pprzros, while also receiving its telemetry. With the help of the Gautier Hattenberger, this framework managed to work. The commands are sent as RC_4CH messages. However, this message channel is 8 bit, and the drone commands are between -9600,+9600, making this solution have low resolution and result in not optimal reference commands and performance. 

As advised in the gitter, I tried to change the message type to int16 and change the scaling and type in rc_datalink.c/.h. But the problem is deeper than that, as the input2ivy is scaling the values down to 8bit as explained in this warning found in messages.xml joystick section "Standard joystick axis values are on 16 bits signed integers, but tools like 'input2ivy' may scale them on int8 type.".

There is any workaround for this? I believe that the pprzros is a very nice tool since it allows research already based on ROS to be very easily integrated into paparazzi. 

Thanks in advance.

gautierhattenberger, 2020-11-07:
This should be made configurable, but as discussed on gitter channel, this is not so easy. So this won't be done soon.

OpenUAS, 2020-11-14:
Don't know about pprzros but how about (Mis)using 2 channels and re-combine those in the aircraft code again as one?

closed 2022-11-25 by OpenUAS