paparazzi/paparazzi #2603: Paparazzi takes too long to load in Ubuntu 20.04

hendrixgr, 2020-10-16:
I am not sure if the problem is with the paparazzi software or Ubuntu 20.04 is to blame but it takes 15-20 seconds in order for the Paparazzi Center to load, if i run the paparazzi with sudo it loads up in a flash!
Running ./paparazzi.sh from a console inside the /home/paparazzi/ directory does not show any warnings or messages, the cursor disappears for 15-20 seconds and then the paparazzi Center appears.

gautierhattenberger, 2020-10-18:
This is most likely a bad Ubuntu/GTK2/paparazzi interaction, but there is not much we can do unfortunately. For some people, the problem is with the paparazzi center, and for others it is the GCS.
I seems that using Kubuntu instead of Ubuntu, is slightly better.

OpenUAS, 2020-11-12:
On a clean default Ubuntu 20.04 64Bit install I had success with:

`sudoapt install appmenu-gtk2-module`

After a reboot, no delay anymore with Paparazzi-Center :smile: 

Reopened, just to see if we add this to the install on-liner for time being?

gautierhattenberger, 2020-11-12:
good catch, seems to work for us as well :+1: 

will add that to the next paparazzi-dev package

gautierhattenberger, 2021-03-12:
done with paparazzi-dev 3.28

closed 2021-03-12 by gautierhattenberger