paparazzi/paparazzi #2607: DShot Protocol debugging

aestebanluengo, 2020-10-29:
@OpenUAS , @EwoudSmeur and I are trying to make the Dshot Protocol work on Ewoud's Cyclone hybrid MAV which mounts the Chimera board with the code created by @alex31 and maintained by @gautierhattenberger . The aim is to log the RPM of both motors for system identification purposes.

The branch in which we are working is the following: [https://github.com/aestebanluengo/paparazzi/tree/cyclone_sid](https://github.com/aestebanluengo/paparazzi/tree/cyclone_sid)

The RPM data is logged to the SD card using the following script: [https://github.com/aestebanluengo/paparazzi/blob/cyclone_sid/sw/airborne/firmwares/rotorcraft/autopilot_static.c](https://github.com/aestebanluengo/paparazzi/blob/cyclone_sid/sw/airborne/firmwares/rotorcraft/autopilot_static.c)
Check lines 150 and 151:
```
sd_buffer_i[11] = esc_telem->rpm;
sd_buffer_i[12] = index_last_telemetry;
```
`esc_telem->rpm` is the electrical RPM value and `index_last_telemetry` is the motor from which the RPM vale is received. As specified in the airframe file ([https://github.com/aestebanluengo/paparazzi/blob/cyclone_sid/conf/airframes/tudelft/cyfoam.xml](https://github.com/aestebanluengo/paparazzi/blob/cyclone_sid/conf/airframes/tudelft/cyfoam.xml))
```
<servos driver="DShot">
    <servo name="RM" no="0" min="0" neutral="100" max="1900"/>
    <servo name="LM" no="1" min="0" neutral="100" max="1900"/>
    <!--<servo name="SERVO_TEST" no="1" min="0" neutral="0" max="1900"/>-->
</servos>
```
index = 0 corresponds to the right motor and index = 1 corresponds to the left one.

Also, according to the previous logger indexes, the 12th and 13th columns of the Matlab matrices presented down below correspond to the `esc_telem->rpm` and `index_last_telemetry` values.

The function in which the motor index is specified to send the RPM value corresponds to `dshotSendFrame` in [https://github.com/aestebanluengo/paparazzi/blob/cyclone_sid/sw/airborne/arch/chibios/modules/actuators/esc_dshot.c](https://github.com/aestebanluengo/paparazzi/blob/cyclone_sid/sw/airborne/arch/chibios/modules/actuators/esc_dshot.c):

Original code (the one in my repo is a bit modified from the one presented below whose source is [https://github.com/ewoudsmeur/paparazzi/blob/cyclone_dshot/sw/airborne/arch/chibios/modules/actuators/esc_dshot.c](https://github.com/ewoudsmeur/paparazzi/blob/cyclone_dshot/sw/airborne/arch/chibios/modules/actuators/esc_dshot.c))
```
void dshotSendFrame(DSHOTDriver *driver)
{
  if (driver->dmap.state == DMA_READY) {
    if ((driver->config->tlm_sd != NULL) &&
        (driver->dshotMotors.onGoingQry == false)) {
      driver->dshotMotors.onGoingQry = true;
      const uint32_t index = (driver->dshotMotors.currentTlmQry + 1) % DSHOT_CHANNELS;
      driver->dshotMotors.currentTlmQry = index;
      setDshotPacketTlm(&driver->dshotMotors.dp[index], true);
      /*driver->dshotMotors.dp[index].telemetryRequest = 1;*/
      chMBPostTimeout(&driver->mb, driver->dshotMotors.currentTlmQry, TIME_IMMEDIATE);
    }

    buildDshotDmaBuffer(&driver->dshotMotors, &driver->dsdb, getTimerWidth(driver->config->pwmp));
    dmaStartTransfert(&driver->dmap,
                      &driver->config->pwmp->tim->DMAR,
                      &driver->dsdb, DSHOT_DMA_BUFFER_SIZE * DSHOT_CHANNELS);

  }
}
```
According to the previous code, the variable `index` should toggle between 0 and 1, corresponding to the two different motors.

When the `index` variable is allowed to toggle and the DShot speed is specified to 600, the following happens:
- When the index = 1, sensible RPM values are obtained when the motors are off and when they are running.
- When the index = 0, the RPM is fixed to random and unrealistic number equal to 31041.
- The toggling pattern is irregular when the motors are off, sometimes two consecutive 1's appear for the index while others only a single value is read.
- The toggling pattern is regular when the motors are running, two consecutive 1's followed by a 0.

[image]
[image]
[image]
The previous plot looks appear filled in blue due to the large and constant RPM values obtained when the index = 0.

When the `index` variable is allowed to toggle and the DShot speed is specified to 150, the following happens:
- When the index = 1, sensible RPM values are obtained when the motors are off and when they are armed.
- When the index = 0, the RPM is fixed to random and unrealistic number equal to 14657.
- The toggling pattern is always regular with two consecutive 1's followed by a 0.

[image]
[image]

Now, if the `index` variable is hardcoded to 1, the `index_last_telemetry` values do toggle with the same pattern as in the previous case, both in terms sequence and in terms of RPM values obtained for each motor index. Toggling in this case should not occur.

Finally, if the `index` variable is hardcoded to 0 and the DShot speed is specified to 600, the `index-last_telemetry` values are always 0 (as they should, no toggling occurs) and the RPM values obtained are sensible except for the peak obtained when the throttle passes from idle to a higher value. The RPM at the peak is 35153.

[image]
[image]
[image]

If the `index variable` is hardcoded to 0 but the DShot speed is lowered to 500, no peak occurs:
[image]

The solution we are looking for is to obtain sensible RPM values from each motor with a constant toggling pattern in terms of motor index of 0-1-0-1 and so on.

alex31, 2020-12-12:
Hello, 
@aestebanluengo, and sorry for the delay, i've been quite buzy.
The first problem with dshot telemetry is we do not use it @enac, so it has not been tested. Second problem is, in this locked period, i won't have hardware to test fixes.

I looked over the source and found something stupid : the channel toggling is done in the sending function and not in the receving function. I think all the state machine is flawed and need rewrite. 

dshot frame are sent at high frequency, but telemetry frame are received at low frequency, so I have to modify code so that channel toggling be done in the telemetry receive function, but with a timout management : telemetry demand can be discarded by an esc.

I won't be able to test that code, so the burden will fall on you for the tests. If you agree to be beta tester, i can work on it next week.

fvantienen, 2021-01-05:
Hello @alex31 ,

The master student working with this isn't here anymore and unfortunately can't test anymore. We are still interested in this feature in the future, but don't need it right now.
I'm currently looking into bidirectional dshot, which could be more interesting to implement and also doesn't look that hard. It has a lot of advantages over this original dshot telemetry.

Greetings,
Freek

alex31, 2021-01-05:
Hello Freek,

Ok, I  leave that fix in my to-do list, here in the lab, Murat
seems interested to have it functional eventually.

For the bidirectional dshot, that is interesting, and if you are
looking at it, I'm interested in your thought !

The problem that I see with bidirectional dshot is we use in current
autopilots, the 4 channels of 2 timers to get 8 pwm (dshot or
conventional) outputs. 

Unfortunately, there is a hardware limitation on the STM32 timer, input
capture is only possible on the first 2 channels, so won't be able to
use the timer in DMA for the telemetry reading of channels 3 and 4. What
is your plan to read the telemetry on these channels, bitbanging ?

If designing a new autopilot using only the first two channels of 4
timer to get 8 input/output, it would be relatively easy to reconfigure
timers and read the channels (not the two together, it's not possible,
but one after the other) in DMA to get the telemetry.

Also, I am interested to test if it's possible to use the stm32H7 mdma
linked list feature to automatically chain timer dshot output and
telemetry input, reconfiguring the timer in the transfer complete ISR.

Greetings,
Alexandre

Le Tue, 05 Jan 2021 05:14:57 -0800,
Freek van Tienen <notifications@github.com> a écrit :

> Hello @alex31 ,
>
> The master student working with this isn't here anymore and unfortunately can't test anymore. We are still interested in this feature in the future, but don't need it right now.
> I'm [...]

-- 

Alexandre

fvantienen, 2021-01-06:
Hello Alexandre,

Do you have any documentation on why the IC only works on the first two channels? Since I think both betaflight and ardupilot are capable of having bidirectional dshot with 4 channels (at least as far as I checked myself).

I want to implement this on the px4fmuv5, where it should be possible according to ardupilot. Not sure about your hardware?

Greetings,
Freek van Tienen

fvantienen, 2021-01-06:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/RCOutput_bdshot.cpp

alex31, 2021-01-06:
Le Wed, 06 Jan 2021 08:03:37 -0800,
Freek van Tienen <notifications@github.com> a écrit :

> https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/RCOutput_bdshot.cpp
> 

ChibiOS input capture driver (ICU) is in fact a PWM Input mode, this
mode capture the width and the period of a signal. This mode is indeed
limited to 2 channels (1 and 2)

To read DSHOT telemetry, one do not need PWM input since the period is
always the same, only the width is useful, so one can
use input capture mode (for wich there is no ChibiOS driver), and input
capture mode permits 4 channels.

Which ESC have you tested Ok for telemetry dshot ?

-- 

Alexandre

fvantienen, 2021-01-15:
I have some BLHeli32 ESC's which are now outputting telemetry, but haven't got the input DMA working though. You have to invert the PWM pulses and also the CRC, and based on that it will start sending telemetry. Not sure though why my current DMA input setup is not working, got to do some debugging.

The problem with the current setup is that we have to do the same trick as Ardupilot, that we have to pull them 1 by 1. Since we are using the PWM_UP dma stream, instead of the separate channel streams betaflight is using. Which is fine for now 👍

Fabien-B, 2024-04-12:
Closing this issue: bidirectionnal DShot is implemented, and a lot of things have changed since.
Re-open it if needed.

closed 2024-04-12 by Fabien-B