paparazzi/paparazzi #2608: fatal error: nav_cube.h: No such file or directory

jburns11, 2020-11-03:
Hello all,

I ran into an error when trying to use nav_cube.h. If you insert either of the lines 
<module name="nav" type="cube"/>
or
<module name="nav_cube"/>
into an aircraft file, you will get an error from settings.h. This file tries to include nav_cube.h, but from what I understand it should be trying to include nav/nav_cube.h.

It seems like when settings.h is being generated, the directory tags are not being correctly parsed from the aircraft file. For instance, when I include the "nav_smooth" module in the aircraft, the import statement is nav/nav_smooth.h.

Best,
Jack

gautierhattenberger, 2020-11-03:
You're right, there is a mistake in the module xml file. Every `module="nav_cube"` in the `dl_setting` nodes should be `module="nav/nav_cube"`.
This is a pain for now, but want to fix this someday so that these includes are automatic. A pull request to fix it would be greatly appreciated :)

jburns11, 2020-11-03:
Created a PR for this! Guess this issue can be closed!

commit 3ced6ffe (2020-11-03): Fixed paths in nav_cube.xml module file. Also fixed setter macros in nav_cube.h for offset values (#2609)

close  #2608
closed 2020-11-03 by gautierhattenberger