paparazzi/paparazzi #2611: ChibiOS Deployment with Paparazzi

iAbdur18, 2020-11-04:
Hi Everyone, Hope you find this message in good health and spirit,

I am a newbie and need to deploy ChibiOS/RT with Paparazzi from scracth with few models to show, can anyone guide me on how to begin and which documents to read?

Regards
Abdur Rehman

gautierhattenberger, 2020-11-07:
Hi,
We don't have a proper guideline for integrating new boards with ChibiOS. You can have a look at the pull request as an example: https://github.com/paparazzi/paparazzi/pull/2481
You can also check the files in this directory: https://github.com/paparazzi/paparazzi/tree/master/sw/airborne/boards/tawaki/chibios/v1.1
The `tawaki_1.1.cfg` is a configuration file that can generate a ChibiOS `board.h` file using this tool: https://github.com/alex31/chibios_enac_various_common/blob/master/TOOLS/boardGen.pl (see doc: https://github.com/alex31/chibios_enac_various_common/blob/master/TOOLS/DOC/boardGen.pdf)

iAbdur18, 2020-11-10:
Hey @gautierhattenberger , thanks for the replay and I have read your guidelines and will try to get it accordingly. Thanks for the help

gautierhattenberger, 2020-11-12:
Closing this issue, but tagging it to the documentation project as this should be described somewhere

closed 2020-11-12 by gautierhattenberger