paparazzi/paparazzi #2621: Logplotter crashes on valueplot on ubuntu 20.04

OpenUAS, 2020-11-20:
Start logplotter, take a log generated, try to plot a value,  crashes logplotter regardless which message chosen

e.g. via

```
 cd ~/paparazzi/sw/logalizer
 ./logplotter
```

Seem to involve units.xml 

` Unit conversion error: Parse error of conf/units.xml`

Many lines like these output on console, likely number amount of lines same as number of messages types in the logfile
Seem to be related to changes to use PprzLink of 2016  like +  | (m, values)::_ when m.PprzLink.name = "GPS" ->
Tried to simplify the units.xml in own conf.. no difference. Tried on 3 different PC's with PPRZ and PPRZ 20.04  all the same crash.

gautierhattenberger, 2020-11-20:
Can you tell me more about the log file: is it an old log or a recent one ? from real flight or simulation ? from the same machine or not ?

Fabien-B, 2020-11-20:
I reproduced the bug on 20.04 with both log from real flight and simulation. However I don't get your error, I just get `Segmentation fault (core dumped)`.

gautierhattenberger, 2020-11-20:
also, did you define the `PAPARAZZI_HOME` env variable ? eventually do you have also a `PPRZLINK_DIR` ? And also, do you have the `unit.xml` file in the `var` folder of paparazzi ?

Fabien-B, 2020-11-20:
I tested, I think that the errors `Unit conversion error: Parse error of conf/units.xml` are due to not having the PAPARAZZI_* environment variables sets.
But crash with `Segmentation fault (core dumped)` anyway.

gautierhattenberger, 2020-11-23:
@OpenUAS you can try #2623 as a workaround for now (core dump issue).
The initial issue about parse error should be solve by setting env variables properly.

closed 2020-12-03 by Fabien-B

OpenUAS, 2020-12-11:
little late, anyhow it worked :) thx, lots of happy students now also..