paparazzi/paparazzi #2642: HGT not download anymore

OpenUAS, 2021-01-06:
A directory server change on https://dds.cr.usgs.gov/ make current OCAML code not download SRTM anymore
continents are now in subdir e.g. https://dds.cr.usgs.gov/srtm/version2_1/SRTM3/Australia/
A way to change the code could be try to download from first in a list ,if download not possible try next of list since the list is small easier to code and take less space than a lookup table which file belon to which directory... an likly just as fast...

Africa/
Australia/
Eurasia/
Islands/
North_America/
South_America/

OCAML Fu is missing at the moment, else would have made PR, for now for the needy ones (with HD space left ;)

Run this oneliner for your continent ;)

gautierhattenberger, 2021-02-08:
I can't access link above without a login

dewagter, 2021-02-10:
What about using the ESA server: everything in a single folder.
 - http://step.esa.int/auxdata/dem/SRTMGL1/

gautierhattenberger, 2021-02-10:
I like the idea :) thanks

PR #2659 (2021-02-11): [ocaml] fix SRTM download path
SRTM data are no more available for direct download, using ESA data
instead. The new SRTM data are SRTM3 GL1 (1 arcsec instead of 3
previously), so the code is now adapted to support both formats,
the most accurate available will be used first.
The only way to force the update of new files is to remove the previous
ones from data/srtm folder.

close #2642
commit 80b2c91a (2021-02-11): [ocaml] fix SRTM download path (#2659)

SRTM data are no more available for direct download, using ESA data
instead. The new SRTM data are SRTM3 GL1 (1 arcsec instead of 3
previously), so the code is now adapted to support both formats,
the most accurate available will be used first.
The only way to force the update of new files is to remove the previous
ones from data/srtm folder.

close #2642
closed 2021-02-11 by gautierhattenberger