paparazzi/paparazzi #2673: LDFLAGS order reversed

tomvand, 2021-03-18:
Hi, the current version of paparazzi seems to reverse the order of the LDFLAGS, which are order-dependent and this causes problems when linking multiple dependencies such as in the [cv_opencvdemo module](https://github.com/paparazzi/paparazzi/blob/master/conf/modules/cv_opencvdemo.xml). This problem did not occur during last year's tu delft pprz course, so apparently something has changed in the meantime.

gautierhattenberger, 2021-03-18:
Do you have an idea since when you have this problem ?

tomvand, 2021-03-18:
No. I only discovered this problem today, while I haven't used this module (or anything else with complex ldflags) since the course one year ago. But I realize that's quite a large timespan...

gautierhattenberger, 2021-03-18:
It is possible that the final order is currently reversed compared to the original order in module XML. Can you check that ?

closed 2021-04-08 by gautierhattenberger

tomvand, 2021-04-20:
Sorry for not responding, I was ill. Thanks for the fix!

PR #2684 (2021-04-08): [build] output flags in orginal order
it does matter for LDFLAGS in particular

close #2673
commit 2cd3c38d (2021-04-08): [build] output flags in orginal order (#2684)

it does matter for LDFLAGS in particular

close #2673