paparazzi/paparazzi #2714: EKF2 waypoint altitude not correct

gautierhattenberger, 2021-05-06:
It seems that the altitude of the waypoint is always reset above 0 hmsl instead of above ground level (at least in simulation) when using EKF2 ins filter. I have the same result with and without the fix #2713 .
Is it something that is also happening in real flights ?

fvantienen, 2021-05-06:
I don't fully understand the problem, but is it that the `ekf2` `origin` is always set to 0 hmsl?
The `ekf2` doesn't support origin initialization from the flight plan, this is because the default interface from PX4 doesn't support setting the origin. If this is the problem you are talking about then I can take a look on how to fix it?

gautierhattenberger, 2021-05-06:
I guess that is the problem, I just saw this when running a simulation. I don't think it is a problem to have the INS working in its own frame, but that would be better if the pprz wrapper moves the inputs and outputs to get the correct altitude in the GCS.

gautierhattenberger, 2022-10-27:
close with #2829 and #2930

closed 2022-10-27 by gautierhattenberger