paparazzi/paparazzi #2715: EnuCoor_i and NedCoor_i units

Fabien-B, 2021-05-06:
EnuCoor_i structures can be expressed in INT32_POS_FRAC or centimeters, and its not always clear which unit is used.
I think its a huge risk to introduce bugs.
I would suggest reserving this struct for INT32_POS_FRAC coordinates, and make a new structure EnuCoorCm_i if centimeters are needed, to reduce the risk of errors.

Does anyone have an other suggestion ?

fvantienen, 2021-05-06:
I didn't know this problem existed. But seems like a good solution.
For me personally I prefer to do most calculations in float, since almost all autopilots can easily compute floats.

fvantienen, 2021-05-10:
I was not meaning to really rewrite most of the code, but more if new code is made or code is refactored I don't mind anymore if it is only made using float's. Since it avoid's a lot of error's and mistakes with converting and precision etc.