paparazzi/paparazzi #2726: Battery number of cells

fvantienen, 2021-05-19:
Currently we are flying with a lot of different platforms with different batteries and number of cells. For me (and I guess also for others) it would be really useful if you could define `BAT_NUMBER_OF_CELLS` inside the `BAT` section of your airframe and the battery strip inside `strip.ml` would divide automatically by this number.

I tried to add this myself in ocaml, but came as far to forward to the `strip_param` and update the max/min values but couldn't find an easy way to get it inside the `set_bat` callback of `strip.ml`. Could you maybe give me a pointer on how to do that part correctly?

Fabien-B, 2021-05-19:
Sorry for the "out of context" message, but if you have some time, you can test the new GCS I am currently working on, made with C++/Qt : https://github.com/Fabien-B/PprzGCS
This is still a work in progress, and the battery management is not implemented at the moment, but it will be soon.
I know this is not helping right now, but the current GCS use deprecated software, that Gautier rebuild himself just for the GCS, so its end of life is really approaching...

gautierhattenberger, 2021-05-19:
@fvantienen how do you get the battery number ?

fvantienen, 2021-05-19:
Just defining in the `BAT` section of the airframe, like the `MAX` and other voltages.
Something like this:
`fvalue "CATASTROPHIC_BAT_LEVEL" default_catastrophic_level, fvalue "MAX_BAT_LEVEL" default_max_level, fvalue "BAT_NUMBER_OF_CELLS" default_number_of_cells`

gautierhattenberger, 2021-05-19:
sorry, I misunderstood, I thought you wanted to monitor several batteries. So you want to have the voltage / cell instead of total voltage, or to have both ? It could be included in a tooltip as well.

fvantienen, 2021-05-19:
Yes indeed I just wanted to have by default show the 'voltage / cell', because that is similar for all drones. Now we are usually calculating by hand every time we fly. Wanted to make it such that if you don't define it or set it to `0` it would show as previous.

I would prefer it in the main view instead of a tooltip, but both are fine.

closed 2021-05-21 by fvantienen