paparazzi/paparazzi #2730: Guidance_h ref acceleration jumps

EwoudSmeur, 2021-05-20:
When simulating tudelft/bebop_indi_actuators.xml, trying to include the reference acceleration from the horizontal reference as a feed forward, I noticed that it was jumping to zero and back (jittering) all the time when going to a new reference position and the speed is saturated. I traced the problem to the function gh_saturate_ref_speed (https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/guidance/guidance_h_ref.c#L144). Or it could be the function that calculates the max speed based on the heading angle the drone (https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/firmwares/rotorcraft/guidance/guidance_h_ref.c#L141).

I think this affects any controller that uses the reference acceleration. I think I can solve this bug, but I'm wondering if we are happy with the current reference model or if we want to change it (3D, floating point?), before I spend time on it.

OpenUAS, 2021-09-22:
Thanks for reporting and offer to fix. @gautierhattenberger any opinion on this welcomed.

gautierhattenberger, 2021-09-22:
If we can have a smooth acceleration setpoint, I think it is better than the current situation even if it "works". I also don't mind that much if the internal computation is done in floating point, and even better if it is a more generic library.

closed 2022-11-28 by EwoudSmeur