paparazzi/paparazzi #2755: Unexpected behaviors on the nps

KimHyungSub, 2021-07-22:
When I tested several frames on the nps (JSBSIM) simulator, I noticed several unexpected behaviors.
I am not sure whether these are unknown issues or not. Just in case, I would like to leave the issues in here.

1) 
- Vehicle type: LadyLisa
- The following flight mode change leads to increasing altitude.
---> Hover mode -> Hover_Z or Hover_C
---> The vehicle keeps increasing its altitude during Hover_Z or Hover_C mode.

2) 
- Vehicle type: bebop2
- The vehicle show divergence of X and Y positions after conducting FLIP mode.

3) 
- Vehicle type: bebop2, ardron2
- Many flight modes (e.g., Hover and ATT) are not triggered (maybe, bebop2 does not support the modes).

4) 
- Vehicle type: ardron2
- Losing attitude control and then crashing on the ground after FLIP flight mode

5) 
- Vehicle type: Quad-Navstik
- Fail to takeoff

6) 
- Vehicle type: MentorEnergy
- FAILSAFE never be set even though explicitly triggering FAILSAFE mode.
- Vehicle type: Quad_Lisa_2, Quad_LisaMX, Minion_Lia
- When I disconnected AHRS, INS, and GPS, the FAILSAFE can triggered only if its flight mode is NAV. In other words, if the vehicle is in other modes such as Hover_C and Hover_Z, the vehicle never trigger the FAILSAFE. But, the Hover_C is still requiring position data from GPS or INS.

- These issues maybe come from my misunderstanding. Please correct me if I am wrong. 
- If these issues are bugs, it would be great if these are fixed :) Thanks!

OpenUAS, 2021-08-19:
Thanks for testing and reporting, really appriciated. A log file of your simulated flight attached to this issue would help into determining several causes. Then it is also clear which PPRZ version of better GIT object ID(SHA) was at the time of testing

OpenUAS, 2021-09-22:
@KimHyungSub thanks for adding the Git object ID (SHA) that is already a good step to investigate the issue reported.

KimHyungSub, 2021-09-23:
Hello @OpenUAS,
Thanks for your work!