paparazzi/paparazzi #2757: GCS waypoints too small to select on HDPI screens

EwoudSmeur, 2021-08-05:
On higher resolution screens the waypoints become very small and hard to select/drag, and the drone icon is also so small that its heading is not really visible. Can we make this depend on the screen resolution or the Ubuntu display 'scale' parameter?

gautierhattenberger, 2021-08-06:
This is a big effort for something that is using a deprecated library (gtk2). We are now working on a new GCS https://github.com/Fabien-B/PprzGCS and if possible don't change things in the old one.

EwoudSmeur, 2021-08-09:
Very cool, I just installed it and it has a completely different look and feel! I'll be using this GCS for the coming flight tests. Would you like me to report any issues I encounter? e.g. pinch to zoom works on my touchpad, but not on the touchscreen :)

gautierhattenberger, 2021-08-10:
Of course, don't hesitate to report any issue and even wished features, we will consider that.

Fabien-B, 2021-08-25:
@EwoudSmeur I implemented multitouch support in latest release! Pinch to zoom and pan.
If you have further issues/whished features, please report them here: https://github.com/Fabien-B/PprzGCS/issues

EwoudSmeur, 2021-08-25:
That is so cool, thank you! 😍
I'll let you know when I find something!

Fabien-B, 2021-08-25:
I am curious: did you install it source sources or did you used the .deb? On what OS?

closed 2021-08-30 by Fabien-B