paparazzi/paparazzi #2781: Magnetometer RM3100 crashing after some time

matteobarbera, 2021-09-22:
I have attempted to connect an RM3100 magnetometer to a Pixracer, but I have encountered problems with it crashing (it stops sending messages) after a few minutes (sometimes seconds) of being powered on.

In order to get it to compile I had to add i2c dependency to https://github.com/paparazzi/paparazzi/blob/17422e468bfd5633f0453bcda76237b3fc7168ed/conf/modules/boards/px4fmu_4.0_chibios.xml#L10
which perhaps would be useful to change on master, maybe together with the rest of the px4 boards?

Originally I thought the issue was hardware related, but @OpenUAS also encountered the same issue with another RM3100 sensor (and different board?). Using RM3100_RATE_75 seems to not cause a crash.

OpenUAS, 2021-09-22:
Indeed, same issue, that is on master as of 18092021 (sorry, no sha at hand) before I've tested worked 100% at least for 30 minutes on Lisa MX last test on Lisa MXS but unlikely the couse.  Need to take a look at pullups and module vcc voltage ripple. Will investigate with @matteobarbera tomorrow.

Yes, a change to <depends>i2c, spi_master,baro_board</depends>   works better ;)

gautierhattenberger, 2021-09-22:
It seems that the `dep` node is missing for this sensor, and it should be
```
<dep>
  <depends>i2c</depends>
  <provides>mag</provides>
</dep>
```

matteobarbera, 2021-09-23:
After some more investigation we couldn't find any issues in the driver or other software, and the current hypothesis is that the current sensor is acting up and that the autopilot is not getting a consistent current, as we saw spikes in the sensor data that stopped after fiddling with the power supply. We will try to install a new current sensor and a dedicated BEC for the magnetometer.

A potential issue that we discovered, however, is that the data is only ever polled at the module frequency https://github.com/paparazzi/paparazzi/blob/5dd484ec8dde34af89979410527f82c6b83ea230/conf/modules/mag_rm3100.xml#L29 This means that if the rate of the sensor is set higher (maximum configurable sensor frequency is 600 Hz) the autopilot only ever reads the data at 60 Hz, which makes this high frequency magnetometer somewhat pointless and may even cause issues with the sensor buffer. I don't know whether to set the module frequency to the max autopilot periodic frequency or simply set the default to be higher.

OpenUAS, 2021-10-04:
re-tested with RM3100 with stable own power-level converter on the RM3100 module for 25minutes all is OK.

@matteobarbera Your remark about "mag_rm3100_module_periodic()" freq="60"/>  still holds, but that should be a separate issue.

closed 2021-10-04 by OpenUAS